RobotLocomotion/drake #15696: temp_directory() has ambiguous semantics over directory ownership

SeanCurtis-TRI, 2021-09-01:
# Problem

If a user calls `drake::temp_directory()` they are provided a string representing the path to a directory. The actual path depends on a number of things (see below), but the one thing the caller *cannot* know is whether they are responsible for cleaning up that directory or not. And, in fact, the responsibility to clean up typically depends on whether the code is running in a testing context or running as a regular binary (but that same mechanism could be used in other contexts).

A piece of code that creates temporary files and is responsible for cleaning them up cannot reliably take that action in the real world *and* in unit tests without going through absurd code gymnastics.

# Details

The logic for `temp_directory()` works based on cascading logic of examining environment variables:

  1. If `TEST_TMPDIR` is defined, then
        - the path indicated by that variable's value *must* already exist.
        - It must be a directory.
        - That directory *is* the return value.
        - The caller is not responsible for deleting it (and probably can't).
  2. Else, look for a "temp" prefix:
        - If the environment variable `TMPDIR` is defined, use that as a prefix.
        - Otherwise, simply use `/tmp`.
  3. Affix a temp directory  name `robotlocomotion_drake_XXXXXX` to the prefix (and create the corresponding directory).
        - Return the `prefix/robotlocomotion_drake_XXXXXX` path.
        - The caller *is* responsible for deleting this folder (as it was created as a direct result of the caller's action).

# Proposed Solution(s)

The most obvious solution is to treat `TEST_TMPDIR` just the same as any other prefix. `Drake` doesn't write directly into that directory, but creates its `robotlocomotion_drake_XXXXXX` sub-directory that the caller is always responsible for cleaning up. For unit tests, where bazel provides the temp directory (and defines `TEST_TMPDIR`, knowing that bazel will do the clean up and taking no explicit clean up action is sufficiently "responsible".

A lesser option would be to change the return type into a `std::pair<std::string, bool>` such that the `bool` communicates the need for caller cleanup -- but that would mess up a lot of call sites.

(BTW There's no really suitable "component" label for this kind of thing. Votes?)

jwnimmer-tri, 2021-09-02:
The intent of `temp_directory()` is that the application code _never_ cleans it up.  Either it was running under bazel test, in which case bazel will remove it, or it was a non-test user application in which case the directory persists until the next reboot (par for the course, for `/tmp`).

When running under bazel test, if we want to change the implementation to use a subfolder of `$TEST_TMPDIR`, then I have no problem with that.

What I would find more problematic is changing the documentation of `temp_directory()` to require or imply that the caller has a responsibility of removing the directory when done.  The function intended to be convenient to use.  Keeping track of the responsibility to `rmtree` later is much harder to use.  We could _clarify_ the documentation to help prevent the doubts mentioned above, but the typical use should remain as "caller does not rmdir".

SeanCurtis-TRI, 2021-09-02:
Thanks for looking.

It seems that we're basically on the same page. I feel that the statements

> That decision about deletion would be up to the caller, based upon the conditions where they are using it.

and 

>  the caller is always responsible for cleaning up. For unit tests, where bazel provides the temp directory (and defines TEST_TMPDIR, knowing that bazel will do the clean up and taking no explicit clean up action is sufficiently "responsible".

are philosophical cousins.

So, what this will come down to is making sure we get the language in the documentation correct. What does it mean to be "responsible" for the new directory?

  - In tests, it'll get cleaned up automatically and no action is required from the caller.
  - If the prefix is `/tmp` it'll get cleaned up at the next reboot and, again, no action is required from the caller.
  - Alternatively, the user can delete it themselves as part of clean up.

All of those are examples of the user acceptably exercising its "responsibility". You're quite good at concision, do you have any suggestions on the language?

jwnimmer-tri, 2021-09-02:
Maybe this?

```c++
/// Returns a newly-created directory suitable for temporary files. The caller
/// is not expected to delete the directory, but is free to do so if necessary.
/// ... [explanation of heuristics] ...
```

(I don't think the `@returns` syntax in this docstring is particularly helpful, so I'd probably toss it as well.)

SeanCurtis-TRI, 2021-09-02:
Ok, I've got a draft PR related to this. I'll update the documentation and assign you as feature reviewer. 

I like the emphasis on "Generally, you're under no obligation, but you are fully equipped to do so."

PR #15698 (2021-09-02): [common] Always allow caller clean up directory from temp_directory()
Modify temp_directory so the caller can always clean up.

For code that creates a temporary working directory, this cleans up the ambiguity of whether or not it is able to delete that directory. Now it can *always* clean up (modulo catastrophic system failures).

Updates the tests that confirm the behavior when `TEST_TMPDIR` is set.

resolves #15696

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15698)
closed 2021-09-02 by soonho-tri