RobotLocomotion/drake #15733: Point contact model seems to generate wrong friction force

hongkai-dai, 2021-09-07:
I have a ball-paddle simulation, where the paddle if fixed, and the corner of the paddle is at (0, 0). I drop a ball slightly to the left of the paddle, with initial x position being -5E-4. At the moment of the ball hitting the paddle, the contact force pulls the ball into the paddle, while the ball horizontal velocity is 0. Here is a snapshot of the animation
[image]
I also plotted the horizontal velocity of the ball (ignore the timing, it should start at around 0 but instead 0.4s). You can see that the ball horizontal velocity accelerate in the +x direction first, indicating the frictional force is in the wrong direction
[image]

I have a branch https://github.com/hongkai-dai/drake/tree/ball_paddle_sim_error. You can do
```
$ bazel build //examples/ball_paddle:contact_sim_demo_py
$ bazel run //examples/ball_paddle:contact_sim_demo_py
```
to generate the simulation.

Another interesting fact is that if I remove the friction (setting the value to zero) at https://github.com/hongkai-dai/drake/blob/dbd0e57bd9ca9f2868e2004ef76ed2f3b0904c90/examples/ball_paddle/ball_paddle.cc#L95 and https://github.com/hongkai-dai/drake/blob/dbd0e57bd9ca9f2868e2004ef76ed2f3b0904c90/examples/ball_paddle/ball_paddle.cc#L73, then the contact force points to the right direction, as shown in 
[image]
where the contact force points upward and slightly to the left (which I think is the vector from the corner of the paddle to the center of the sphere).

cc @SeanCurtis-TRI @DamrongGuoy @sherm1

DamrongGuoy, 2021-09-07:
CC: @amcastro-tri .  He might be too busy due to a paper deadline.

sherm1, 2021-09-07:
With the normal as shown in the second image, I believe the ball does have a very small tangential velocity to the left. That should produce an opposing friction force to the right. So qualitatively we would expect the net force at the contact point to point to the right somewhat at first contact. Quantitatively it seems too strong though.

cc @mitiguy

hongkai-dai, 2021-09-07:
> With the normal as shown in the second image, I believe the ball does have a very small tangential velocity to the left. 

@sherm1 , do you mean this image?
[image]
This is for frictionless contact, so the force direction is the normal direction (doesn't depend on the ball velocity). I believe the reason why this force direction points to the left a little bit, is because the contact normal points from the corner of the paddle to the ball center, which lies slightly to the left of the paddle.

SeanCurtis-TRI, 2021-09-07:
tl;dr Over to @amcastro-tri  -- looks like an artifact of computing the tangential component due to friction. Certainly consistent with @hongkai-dai's observation that things clean up when friction = 0.

Here's some quick analysis:

[image]
[image]
[image]

This is the measure of the normal/force acting *on the box*. Specifically, the x-component. Given the center of the sphere is slightly to the left of the box, we'd expect the normal and normal components of the force to always have a *positive* component. They do.

However, the tangential component is massive *and* in the opposite direction.  This seems to suggest that it's a dynamics problem.

------------------------------

NOTE: This data is a bit "blurry". I'm not sure what the default integrator is, but we're getting several redundant evaluations at the same time (so, I presume, one of the error-controlled integrators). These graphs include *all* of the contact evaluations (even the ones that presumably got replaced with a subsequent evaluation).

sherm1, 2021-09-07:
@hongkai-dai yes that's the image I meant. I understood that it was the frictionless case. My point was that you can see from that image that the contact point actually has a small velocity to the left (viewed from the contact frame). I know that's not reflected in the force in that image, but it explains why you would expect the friction force (when friction is turned on) to point to the right (into the block).

sherm1, 2021-09-07:
@SeanCurtis-TRI you wrote
> Given the center of the sphere is slightly to the left of the box, we'd expect the normal and normal components of the force to always have a positive component.

I assume you're referring to the x component of the normal force that you plotted above? From the picture it looks like x is positive to the right. Wouldn't we expect then that the force on the ball would be in the -x direction?

hongkai-dai, 2021-09-07:
@sherm1 I see, let me make sure we are on the same page
1. Just prior to the contact, the ball has horizontal velocity = 0.
2. At the moment of impact, the contact normal points from the paddle corner to the sphere center, hence the contact normal points to the left a little bit.
3. We compute the sliding velocity as the component perpendicular to the contact normal. The velocity of the ball is vertically downward, so the sliding velocity is a small-magnitude vector pointing in the 8'clock direction, namely it points to the left and a little bit downward.
4. The frictional force is in the opposite direction of the sliding velocity, hence the frictional force points to the right. The total contact force hence points to almost northeastern direction.

So as long as the closest feature on the paddle is the corner of the paddle, we can observe this behavior that the friction force "pulls in" the ball into the paddle. This is an artifact of the point contact (by using the deepest penetration point to compute the contact normal).

sherm1, 2021-09-07:
@hongkai-dai yes, but even so I would have expected the friction force to be small -- just enough to arrest the sliding motion of the ball vs the paddle. The ball should develop a small angular velocity so that it has no tangential velocity relative to the paddle at which point it should roll off (or bounce off) to the left. The details would depend on the friction coefficient.

hongkai-dai, 2021-09-08:
@sherm1 I think I understand what happens. As you suggested, the tangential friction force arrest the ball motion.

Here is a plot of the ball x/z position and velocity
[image]
In the bottom-left subplot, I draw the horizontal velocity of the ball. The velocity curve has several segments:
1. Left to A, when the ball is not in contact with the paddle.
2. From A to C, the ball accelerates to the right (+x position).
3. From C to B, the ball de-accelerates a little bit, but still move to the right
4. After B, the ball starts to de-accelerates very fast (namely the ball is under a large force in the -x direction)
Previously we were wondering why the ball has a positive x velocity in the segment AC and CB.

What happens is visualized in this plot. In the segment AC and CB, the closest feature on the paddle is the paddle corner (denoted as `O` which is right at the world origin).
[image]
The contact normal points from the paddle corner `O` to the sphere center `S`. I denote that contact normal vector as `n`, and construct a tangential vector `t` perpendicular to `n`. I draw the ball velocity again, but in the new contact normal/tangential frame (which changes with time as the ball center `S` changes).
[image]
Here I denote the instance `A`, `B`, `C` again, which is the same time instances in the bottom-left plot of figure 1. (where I draw the ball horizontal velocity). You can see that the tangential velocity decreases to zero in the segment `CB`.

So what happens is this
1. At the moment of impact, the ball velocity points vertically downward. Since the contact normal `n` points slightly to the left, the ball velocity has a tangential component (hence `v_t < 0` at moment `A`). This tangential velocity creates a tangential frictional force. Since the friction coefficient is 1, this creates a total contact force almost in the northeastern direction.
2. As the ball drops further, the tangential velocity decreases to zero, which corresponds to the segment `CB`. Now we switch to sticking contact, and the ball velocity is along the direction `SO` (from the ball center to the paddle corner).
3. Finally the ball center `S` is inside the paddle, and the nearest feature on the paddle switch to the left surface. Since the ball still has a downward velocity, this means the ball is "sliding" on the left surface, hence the total contact force points to the northwest direction.

@RussTedrake , this is the kink happens for the point contact when the initial ball x position is close to zero. When the ball initial position is further to the left, the simulation doesn't go into step 2 (when the ball is regarded sticking to the paddle). The kink happens right when step 2 emerges in the simulation.

SeanCurtis-TRI, 2021-09-08:
After reading all this, I quickly ran this plot -- comparing force magnitudes. It's another way of seeing what @hongkai was talking about.

[image]

You can pretty clearly see the events at times A, B, & C. I assume that no one has done @amcastro-tri's recommended normal cone test. I'll quickly do that.

SeanCurtis-TRI, 2021-09-08:
Actually, in retrospect, that graph *is* the normal cone test.

We want to confirm that `||fₜ||₂ ≤ μ(fₙ ⋅ n̂)` holds. The orange dots represent `||fₜ||₂`, and, as @hongkai-dai pointed out `μ = 1`, so `μ(fₙ ⋅ n̂) = |fₙ|` (the blue dots) as plotted in the figure.

So, in the case where `μ = 1`, the previous plot shows that the tangential component *always* lies within the cone.

sherm1, 2021-09-08:
Beautiful presentation and analysis @hongkai-dai and @SeanCurtis-TRI!

I would add that being within the friction cone is still not sufficient to ensure physical reasonableness during stiction. A further check would be on the total energy of the system (PE+KE). The friction phase should never add energy.

hongkai-dai, 2021-09-08:
Good call Sherm.

Here is the total energy during the simulation. It is non-increasing
[image]

SeanCurtis-TRI, 2021-09-08:
The general trend in the conversation seems to be toward: "This is a quirk of the model, correctly implemented." 

The small mass of the ball is what makes the sideways acceleration apparent (as the initial contact force has nothing to do with relative masses and is all about relative kinematics).

Is there anything we can do to forestall that conclusion? Offer an alternative explanation?

sherm1, 2021-09-08:
This seems physically plausible at least. Next question would be how hydroelastics behaves under the same level of scrutiny.

SeanCurtis-TRI, 2021-09-08:
Another interesting wrinkle is, this is all continuous. How is it for discrete?

DamrongGuoy, 2021-09-08:
With a software engineering hat, should we close this issue please?  We are doing the correct implementation of an understandable limit in the mathematical model that we used for point contacts.

(By the way, may I please remind that this is a public forum. We might prefer less mentioning of TRI plan here please.)

hongkai-dai, 2021-09-08:
Thanks a lot for the very insightful discussion in this issue! Closing it as we already understand the simulation behavior.

closed 2021-09-08 by hongkai-dai

amcastro-tri, 2021-09-08:
I still don't agree that the model is giving an "artifact". IMO it is the right solution. Yes, the ball's COM accelerates towards the right and that is correct. Also, the ball is gaining angular velocity, but no one is looking at that?

IMO this is a perfect application of point contact. The corner vs. sphere is actually one of those cases that point contact can approximate appropriately. I feel like it is easy to blame point contact when some result does not match our expectation. But we should be careful. Even though we do know that point contact has limitations, we should not blame all of our accuracy problems on the point contact approximation.

Now, could hydro work for this problem. Of course yes. Is it worth the additional computational effort? probably not? I believe hydro will not provide new physics here.