RobotLocomotion/drake #15735: Extract / split drake to use the integration system only

agalbachicar, 2021-09-08:
**Context**

A downstream project uses `drake::systems::AntiderivativeFunction` and `drake::systems::ScalarInitialValueProblem` to compute the arc length of a trajectory described as a parametric 3D curve. Those classes help with the conversion back and forth between the arc length and the curve parameter.

The project is immerse into a ROS2 workspace and installing drake as a C++ system dependency has already been done via a cmake project wrapper which selects the specific drake release we need. However, we are managing the installation of this dependency as a separate step from `rosdep` because we could not find the solution to install drake as a system dependency.

On the other hand, drake is a big project and has many dependencies which is not great for our use case. By quickly inspecting the code, we would require to use many of the drake/systems/framework and drake/systems/analysis because of transitive dependencies.

We are only using `double` template instantiations.

We are using drake v0.27.0 but I suspect we can easily bump it to a newer version.

**Requirements**

We would like to:

- Use the aforementioned machinery to solve our problem.
- Reduce the amount of code we depend on from `drake`.
- Integrate the module into a ROS2 ecosystem, targeting `foxy` at least and using `rosdep` to install it. Note that the C++ API is required.

I took a look at:

- #12783
- [from_binary documentation](https://drake.mit.edu/from_binary.html)
- [build system, ... board](https://github.com/RobotLocomotion/drake/projects/2#card-39884815)
- https://github.com/RobotLocomotion/debian-drake
- https://github.com/RobotLocomotion/drake-external-examples/tree/main/drake_ament_cmake_installed

I don't think there is something that matches exactly our use case because of our requirements.

Are there any plans to split drake into smaller libraries and integrate them into the deployment ecosystem you already have? Alternatively, do you see another solution to extract, and modify the chunk of code we need to satisfy our requirements?

jwnimmer-tri, 2021-09-10:
> A downstream project uses AntiderivativeFunction and ScalarInitialValueProblem  ...

FYI be aware of #12857, if you haven't seen it yet.

> On the other hand, drake is a big project and has many dependencies which is not great for our use case.

One of our goals for Drake is to ensure that our dependencies are compatible with typical uses in the ROS & ROS2 ecosystems.  For example, we try not to require versions that are newer than what a ROS user would already have, and try not to install our own _vendored_ copies of public dependencies (e.g., Eigen, fmt) that should be coming from the ROS ecosystem instead of Drake.  We try to use Ubuntu's shipped C++ libraries whenever possible.

Could you elaborate further about what dependencies are most problematic?  It's possible that we could work to remove them, or conditionally-remove them, or make them private, or etc.

FYI on a related note, the project https://github.com/RobotLocomotion/drake-ros is underway, in case that would aid in consuming Drake as part of the ROS ecosystem in the future.

> Are there any plans to split drake into smaller libraries and integrate them into the deployment ecosystem you already have?

No.  Every user that has expressed interest in consuming a subset of Drake actually wants a _different_ subset, and not in a way that lends itself some simple pattern of smaller, inter-related C++ libraries.  There is no partition of the code that would yield any realistic satisfaction for requests such as this one.

What would be slightly more realistic would be to offer the ability to exclude dependencies.  Something like "Drake without VTK" or "Drake without LCM", etc.  Turning off a dependency would turn off features that required it (for VTK, that would mean camera simulation; for LCM, that would be networking middleware, etc.).  This ties into my question above, regarding which acute dependencies are unwanted.

I _do_ have some tentative plans in mind to partition our use of VTK, LCM, and a couple of the other more problematic dependencies in a way that would make then into distinct components, so users could easily avoid those dependencies.

> Alternatively, do you see another solution to extract, and modify the chunk of code we need to satisfy our requirements?

As one option, if you use Bazel then you can depend on only and exactly the C++ object code that you need, and then Bazel will compile and build only and exactly the dependencies induced by your request.

In this case, it seems like you want just `@drake//systems/analysis:antiderivative_function` (which already incorporates `:scalar_initial_value_problem`), and so we can ask its dependencies (via `bazel query 'deps(//systems/analysis:antiderivative_function)' | sed -e 's#:.*##g;' | sort | uniq`):

```
@eigen
@fmt
@spdlog
//common
//common/trajectories
//systems/analysis
//systems/framework
```

As another option, a system like https://github.com/google/copybara can help automate maintenance of vendored and subsetted dependencies, in case you wanted to maintain your own fork of some files.

agalbachicar, 2021-09-13:
First of all, thanks for your thorough response.

> FYI be aware of #12857, if you haven't seen it yet.

I'll take a look! Thanks!

> Could you elaborate further about what dependencies are most problematic? It's possible that we could work to remove them, or conditionally-remove them, or make them private, or etc.

Talking only about the required output of bazel query, I believe Eigen in v0.33.0 points to 3.4 and Ubuntu focal brings 3.3 but as you mentioned before they are obfuscated. Other than that, there is nothing preventing me to use the piece of code of drake that I need.

> FYI on a related note, the project https://github.com/RobotLocomotion/drake-ros is underway, in case that would aid in consuming Drake as part of the ROS ecosystem in the future.

I will take a deeper look at it. We might simplify our custom ament shim making use / contributing if possible to drake-ros.

> No. Every user that has expressed interest in consuming a subset of Drake actually wants a different subset, and not in a way that lends itself some simple pattern of smaller, inter-related C++ [...]

Understood. I just want to point out that looking at the output of blaze query (I could reproduce it locally, thanks for the hint!) there is relatively simple set of dependencies that could be split:

- drake/common/trajectories requires drake/common and drake/math.
- drake/math requires drake/common
- drake/systems/framework requires drake/common and drake/math
- drake/systems/analysis requires drake/common and drake/math.

I am not an expert on drake nor I intend to be, but it seems like there is some kind of mapping between the high level per module libraries that wrap all the tiny dependencies correlation to what a high level set of drake modules could be offered. Again, not trying to open a discussion that has years and I fully understand that you intend to keep as is for the time being.

> What would be slightly more realistic would be to offer the ability to exclude dependencies. Something like "Drake without VTK" or "Drake without LCM", etc. Turning off a dependency would turn off [...]

This is an interesting point and if I may, I think users would benefit from having drake-* libraries rather than drake w/o certain module. I believe the first translates into an extensible system rather than a customized system which sometimes might be harder to maintain and configure in multiple build systems.

> In this case, it seems like you want just @drake//systems/analysis:antiderivative_function (which already incorporates :scalar_initial_value_problem), and so we can ask its dependencies (via bazel query 'deps(//systems/analysis:antiderivative_function)' | sed -e 's#:.*##g;' | sort | uniq):

It has been of great help. Thanks for this hint! 

As soon as I have some ready that I am satisfied with, I'll consider copybara to automate the process.

jwnimmer-tri, 2021-09-13:
If possible, I was looking for a bit more information; I might not have been clear about the particular question:

> On the other hand, drake is a big project and has many dependencies which is not great for our use case.

Can you elaborate on which specific dependencies count as "too many" or "too big"?  I can't really reply about subs-etting Drake until I know what the _specific_ problem is that needs to be solved.  Your call-to-action in the top post was thorough, but did not sufficiently identify the specifics of what, exactly, is objectionable as a dependency.

agalbachicar, 2021-09-14:
Sorry for the misunderstanding. I can tell all the things I do not require from drake and bring in considerable amount of data.

- I don't require any of the python bindings. Using drake 0.27.0 release,  the folder lib/python3.8 weighs 108M and there are another 17M in libPythonQt.so
- I don't require anything related to VTK. I think that would remove ~120MB from shared libraries in lib.
- I don't require solvers. Without this, many of the third party libraries won't be required.
- I don't require examples. Using drake 0.27.0 release under share/drake/examples I have 62M.
- I don't require manipulation. Using drake 0.27.0 release, under share/drake/manipulation I have 134M of data. 
- I don't require geometry. I couldn't dig deeper here.
- I don't require multibody. I couldn't dig deeper here.

jwnimmer-tri, 2021-09-14:
Aha!  The thing I didn't realize is that disk footprint was your objective function.  So to be clear, your use case involves situations where disk space is a finite resource, and that is the primary problem?  Prior to the most recent post, my assumption had been that your complaint would be loading possibly-binary-incompatible libraries at runtime (i.e., ODR / ABI problems), or maybe the startup cost of too much symbol resolution, rather than the disk footprint.

As far as the Python bindings footprint, that seems like something you could solve in your packaging / release system by culling or partitioning out the files that you don't need.  The Python code in Drake depends on the C++ code, but the C++ code does not depend on the Python code; thus, the Python code is safe to remove without hurting the C++ features at all.

This is how Ubuntu packaging works, for example.  The upstream project (aka Drake) installs a bunch of files into `$prefix`, but then the packaging system maintainer in Debian or Ubuntu splits those files across multiple binary packages, e.g., `libgtk-3-common` vs `libgtk-3-dev` vs `libgtk-3-0` vs `libgtk-3-bin`.  The upstream project does not predetermine how its installed footprint is partitioned into binary packages; the packager can split it up in accordance with the packaging system conventions.

As far as the large model data files under share/drake/foo, that also seems like something you could cull or partition on your consuming end, instead of the producing end.  In any case, adding more fine-grained shared libraries is unrelated to the model data's disk footprint.

If we made VTK an optional dependency, that would allow you to cull it and/or omit it from the build as well.

What to do with the third-party solvers is more complicated question.  We could probably allow for turning them all off (this is possible already, though unsupported in CI).  On the other hand, they almost are all just stock Ubuntu packages already, so I'm not clear on exactly what the objection is.

agalbachicar, 2021-09-14:
> So to be clear, your use case involves situations where disk space is a finite resource, and that is the primary problem? 

Actually, that's part of the problem. I just need to consume one part that belongs to the core of drake (depends on common, math and some of the systems modules). If I could solely depend on that, the dependency (or dependencies) that a developer in my project would be required to pull are just a few (direct and indirect). Also, they don't have a high impact in their disk size. Now, if I wanted to open the audience of my project to consume it, I would ask them to pull the entire drake and that might introduce some limitations.

For context, I have also briefly explored boost-math because it comes with an integration module that offers similar semantics. However, it has the drawback of pulling *a lot* of libraries and we paused that exploratory effort. It's not just the problem of being available in Ubuntu.

--------------

With your response above I understand that your proposal is to post-process a drake release to satisfy my dependencies. That could be done with, let's say, copybara and then I would just need to release my new package for consumption in my project.

If my previous statement is correct, then I think you should not consider this ticket as a valid reason to do anything with VTK, the solvers or examples because we would proceed with our own release selection.

jwnimmer-tri, 2021-09-14:
> With your response above I understand that your proposal is to post-process a drake release to satisfy my dependencies. That could be done with, let's say, copybara and then I would just need to release my new package for consumption in my project.

My most recent post above was about post-processing the _results_ of the Drake build (i.e., the _binary_ files; the "Assets" of [v0.33.0](https://github.com/RobotLocomotion/drake/releases/tag/v0.33.0)).  Using copybara to fork a few Drake files into your own source tree would be subsetting the _source_ files of Drake.  Two different ideas.

agalbachicar, 2021-09-15:
Yes, understood. Indeed, it was another proposal just release what I need.

Thanks again!

closed 2021-09-15 by jwnimmer-tri