RobotLocomotion/drake #15738: Visualize hydroelastic meshes

SeanCurtis-TRI, 2021-09-08:
## Problem
In the hydroelastic contact model, declared primitives types (e.g., spheres, cyliders, etc) are represented with a tessellated mesh. The amount of tessellation can depend on a configuration parameter and the resulting can deviate from the idealized primitive. Being able to *see* the actual mesh involved in the contact can help users reason about the behavior they observe.

## Solution

Ultimately, we'll allow users to configure visualizers in a way that they can declare they'd like to visualize hydroelastic meshes, where available.

[This branch](https://github.com/SeanCurtis-TRI/drake/tree/PR_visualize_hydro_geometries) contains the "full" solution (as full as can be at any given time). The branch will be rebased frequently and pulling a local copy should bear this in mind.

### The plan
- [ ] The python path
    - [ ] #15707
        - [ ] Move `ConvertVolumeToSurfaceMesh()` out of `drake::geometry::internal` and into `drake::geometry` and bind it.
        - [ ] Bind the new `SceneGraphInspector` API (see below)
            - [ ] Enable `MeshcatVisualizer` (`meshcat_visualizer.py`) to visualize hydro geometry.
- [ ] The C++ path
    - [ ] Add API to `SceneGraphInspector` to introspect the existence of hydroelastic mesh representation.
        - [ ] Enable `DrakeVisualizer` to visualize hydro geometry.
        - [ ] Enable `MeshcatVisualizer` (in `meshcat_visualizer.h`) to visualize hydro geometry. This is *not* defined in the [branch linked above](https://github.com/SeanCurtis-TRI/drake/tree/PR_visualize_hydro_geometries) yet.

jwnimmer-tri, 2022-02-10:
Transcribing some notes from f2f discussions... hopefully I understood it correctly

The most important visualization is the contact patch mesh.
In the typical case, most users prefer to turn off pressure field visualization.

Thus, doing the patch mesh into MeshCat first would be the priority.

jwnimmer-tri, 2023-09-12:
As best I can tell, the only remaining work here is to finish this TODO:

https://github.com/RobotLocomotion/drake/blob/6a5fb5acc19f937e9b08175838c09a6b1abde45d/visualization/visualization_config_functions.cc#L51-L52

jwnimmer-tri, 2023-09-14:
My hypothesis is that #20179 has solved this.  Could either @joemasterjohn and/or @SeanCurtis-TRI please check and confirm, and close if so?

jwnimmer-tri, 2024-11-01:
Closing for lack of response.

closed 2024-11-01 by jwnimmer-tri