RobotLocomotion/drake #15743: Improve the documentation on resolution_hint

hongkai-dai, 2021-09-09:
As discussed on slack https://drakedevelopers.slack.com/archives/C2WBPQDB7/p1631143826024000

Currently in `AddRigidHydroelasticProperties` and `AddSoftHydroelasticProperties`, it explains `resolution_hint` as
```
If the geometry is to be tessellated, it is the parameter that guides the level of mesh refinement. See MODULE_NOT_WRITTEN_YET. This will be ignored for geometry types that don't require tessellation.
```
This documentation is not clear enough. Initially I thought the resolution here means the number of tetrahedrons on the mesh, as the screen resolution means the number of pixels on the screen. But this `resolution_hint` actually means the size of the tetrahedron. We should at least mention that the unit of `resolution_hint` is meters.

DamrongGuoy, 2021-09-16:
PR 15767 was merged.  My role has been completed.  Re-assign to @SeanCurtis-TRI as part of #13314.

rpoyner-tri, 2022-02-08:
Likely endgame here is land #16449, then add xrefs among the hydro guide, parsing reference, and Add*Hydro*Properties() declarations.

PR #16549 (2022-02-09): [doc,geometry,multibody] Add refs to Hydroelastic User Guide
Closes: #15743

The module ... it is written.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16549)
commit ffa7f593 (2022-02-09): [doc,geometry,multibody] Add refs to Hydroelastic User Guide (#16549)

* [doc,geometry,multibody] Add refs to Hydroelastic User Guide

Closes: #15743

The module ... it is written.
closed 2022-02-09 by xuchenhan-tri