RobotLocomotion/drake #15744: Best way to add terrain to a scene

felixcra, 2021-09-09:
Dear Drake team,
I'd like to learn a bit about humanoid controllers and thought of the atlas example as being a good starting point to try out a few things. I was now wondering about how I could change the terrain. You could please give me a high-level description (a list of bullet points would be enough) of what the appropriate way to add a static terrain in drake is?
Many thanks and best regards,
Felix Crazzolara

ggould-tri, 2021-09-13:
Assigning to @sherm1 for delegation.

FYI In the past we have built terrain by decomposing the terrain mesh into triangular "pillars".  However this is not especially performant.  I don't know what the current best practice is for this.

SeanCurtis-TRI, 2021-09-13:
This kind of question is ideally suited for stackoverflow. @felixcra would you be willing to pose your question there with the `drake` tag so that others can likewise benefit from the discussion?

jwnimmer-tri, 2021-11-12:
@ggould-tri has one good answer.

If you just want a flat plane, then we often add a large box geometry to serve as the "floor".

In any case, => stackoverflow.

closed 2021-11-12 by jwnimmer-tri

RussTedrake, 2022-10-30:
not sure if the stackoverflow post happened, but I suspect this is related to https://github.com/RobotLocomotion/drake/issues/16809