RobotLocomotion/drake #15755: Meshcat license notices

jwnimmer-tri, 2021-09-10:
Drake bundles and installs [Meshcat](https://github.com/rdeits/meshcat), but as part of Meshcat's git tree, it bundles a bunch of JavaScript code into a precompiled `main.min.js` file.  For any dependencies that get bundled into that file and require distribution notices as a condition of their copyright grant, we must ensure that Drake redistributes the notice as part of our installation.

@BetsyMcPhail or @mwoehlke-kitware this is relevant to both `pip` work (#1183) as well as Drake's precompiled releases in general.

Please figure out how to audit what `webpack` is pulling in, and how to get a license notice dump from all of its dependencies, and file a Drake pull request that fulfills any redistribution requirements of those licenses.  If any of the dependencies are copyleft, please immediately raise that to our attention.

\CC @rdeits FYI

mwoehlke-kitware, 2021-09-21:
Summary:
```
MIT: 110
BSD-2-Clause: 7
BSD-3-Clause: 5
ISC: 5
Apache-2.0: 4
CC-BY-4.0: 1
```

Details:
```
@discoveryjs/json-ext@0.5.3 (MIT)
@types/eslint-scope@3.7.1 (MIT)
@types/eslint@7.28.0 (MIT)
@types/estree@0.0.50 (MIT)
@types/json-schema@7.0.9 (MIT)
@types/node@16.7.5 (MIT)
@webassemblyjs/ast@1.11.1 (MIT)
@webassemblyjs/floating-point-hex-parser@1.11.1 (MIT)
@webassemblyjs/helper-api-error@1.11.1 (MIT)
@webassemblyjs/helper-buffer@1.11.1 (MIT)
@webassemblyjs/helper-numbers@1.11.1 (MIT)
@webassemblyjs/helper-wasm-bytecode@1.11.1 (MIT)
@webassemblyjs/helper-wasm-section@1.11.1 (MIT)
@webassemblyjs/ieee754@1.11.1 (MIT)
@webassemblyjs/leb128@1.11.1 (Apache-2.0)
@webassemblyjs/utf8@1.11.1 (MIT)
@webassemblyjs/wasm-edit@1.11.1 (MIT)
@webassemblyjs/wasm-gen@1.11.1 (MIT)
@webassemblyjs/wasm-opt@1.11.1 (MIT)
@webassemblyjs/wasm-parser@1.11.1 (MIT)
@webassemblyjs/wast-printer@1.11.1 (MIT)
@webpack-cli/configtest@1.0.4 (MIT)
@webpack-cli/info@1.3.0 (MIT)
@webpack-cli/serve@1.5.2 (MIT)
@xtuc/ieee754@1.2.0 (BSD-3-Clause)
@xtuc/long@4.2.2 (Apache-2.0)
acorn-import-assertions@1.7.6 (MIT)
acorn@8.4.1 (MIT)
ajv-keywords@3.5.2 (MIT)
ajv@6.12.6 (MIT)
browserslist@4.16.8 (MIT)
buffer-from@1.1.2 (MIT)
caniuse-lite@1.0.30001252 (CC-BY-4.0)
ccapture.js@1.0.8 (MIT)
chrome-trace-event@1.0.3 (MIT)
clone-deep@4.0.1 (MIT)
colorette@1.3.0 (MIT)
commander@2.20.3 (MIT)
commander@7.2.0 (MIT)
cross-spawn@7.0.3 (MIT)
dat.gui@0.7.7 (Apache-2.0)
electron-to-chromium@1.3.822 (ISC)
enhanced-resolve@5.8.2 (MIT)
envinfo@7.8.1 (MIT)
es-module-lexer@0.7.1 (MIT)
escalade@3.1.1 (MIT)
eslint-scope@5.1.1 (BSD-2-Clause)
esrecurse@4.3.0 (BSD-2-Clause)
estraverse@4.3.0 (BSD-2-Clause)
estraverse@5.2.0 (BSD-2-Clause)
event-lite@0.1.2 (MIT)
events@3.3.0 (MIT)
execa@5.1.1 (MIT)
fast-deep-equal@3.1.3 (MIT)
fast-json-stable-stringify@2.1.0 (MIT)
fastest-levenshtein@1.0.12 (MIT)
find-up@4.1.0 (MIT)
function-bind@1.1.1 (MIT)
get-stream@6.0.1 (MIT)
glob-to-regexp@0.4.1 (BSD-2-Clause)
graceful-fs@4.2.8 (ISC)
has-flag@4.0.0 (MIT)
has@1.0.3 (MIT)
human-signals@2.1.0 (Apache-2.0)
ieee754@1.2.1 (BSD-3-Clause)
import-local@3.0.2 (MIT)
int64-buffer@0.1.10 (MIT)
interpret@2.2.0 (MIT)
is-core-module@2.6.0 (MIT)
is-plain-object@2.0.4 (MIT)
is-stream@2.0.1 (MIT)
isarray@1.0.0 (MIT)
isexe@2.0.0 (ISC)
isobject@3.0.1 (MIT)
jest-worker@27.1.0 (MIT)
json-parse-better-errors@1.0.2 (MIT)
json-schema-traverse@0.4.1 (MIT)
kind-of@6.0.3 (MIT)
loader-runner@4.2.0 (MIT)
locate-path@5.0.0 (MIT)
merge-stream@2.0.0 (MIT)
meshcat@0.0.1 (MIT)
mime-db@1.49.0 (MIT)
mime-types@2.1.32 (MIT)
mimic-fn@2.1.0 (MIT)
msgpack-lite@0.1.26 (MIT)
neo-async@2.6.2 (MIT)
node-releases@1.1.75 (MIT)
npm-run-path@4.0.1 (MIT)
onetime@5.1.2 (MIT)
p-limit@2.3.0 (MIT)
p-limit@3.1.0 (MIT)
p-locate@4.1.0 (MIT)
p-try@2.2.0 (MIT)
path-exists@4.0.0 (MIT)
path-key@3.1.1 (MIT)
path-parse@1.0.7 (MIT)
pkg-dir@4.2.0 (MIT)
punycode@2.1.1 (MIT)
randombytes@2.1.0 (MIT)
rechoir@0.7.1 (MIT)
resolve-cwd@3.0.0 (MIT)
resolve-from@5.0.0 (MIT)
resolve@1.20.0 (MIT)
safe-buffer@5.2.1 (MIT)
schema-utils@3.1.1 (MIT)
serialize-javascript@6.0.0 (BSD-3-Clause)
shallow-clone@3.0.1 (MIT)
shebang-command@2.0.0 (MIT)
shebang-regex@3.0.0 (MIT)
signal-exit@3.0.3 (ISC)
source-map-support@0.5.19 (MIT)
source-map@0.6.1 (BSD-3-Clause)
source-map@0.7.3 (BSD-3-Clause)
strip-final-newline@2.0.0 (MIT)
supports-color@8.1.1 (MIT)
tapable@2.2.0 (MIT)
terser-webpack-plugin@5.1.4 (MIT)
terser@5.7.2 (BSD-2-Clause)
three-wtm@1.0.0 (MIT)
three@0.132.2 (MIT)
uri-js@4.4.1 (BSD-2-Clause)
v8-compile-cache@2.3.0 (MIT)
watchpack@2.2.0 (MIT)
webpack-cli@4.8.0 (MIT)
webpack-merge@5.8.0 (MIT)
webpack-sources@3.2.0 (MIT)
webpack@5.51.1 (MIT)
which@2.0.2 (ISC)
wildcard@2.0.0 (MIT)
wwobjloader2@4.0.0 (MIT)
yocto-queue@0.1.0 (MIT)
```

jwnimmer-tri, 2021-09-21:
Some web searching shows this as relevant:
https://www.iusmentis.com/computerprograms/opensourcesoftware/license-notices-web-applications/

Also, I just noticed that `dist/main.min.js` already has this:
```js
/*! For license information please see main.min.js.LICENSE.txt */
```

Does that mean we just need to as Robit to commit that yarn-generated file, and then we're done?

jwnimmer-tri, 2021-09-21:
> As previously noted, generating a .tar with all of the relevant licenses seems plausible. Isn't that what we do for other dependencies?

Yes.  However, this is javascript code, which we don't have anywhere else yet -- it's a new circumstance.  If `webpack` has already solved this problem (and seems to me looks like it has), then we should re-use the standard solution, instead of spending time rolling our own bespoke solution that we'll have to maintain over time.

>  npm run build hangs without completing when I try to run it.

Now that we've seen evidence that `webpack` solves this for us, we should open an issue on rdeits/meshcat asking if they (upstream) can solve the license problem for us, by committing the license file as part of their upstream development.  (They should really be doing this anyway.)  Or at minimum, help you open the PR that does it upstream.

We should only roll our own if the upstream approach fails.

mwoehlke-kitware, 2021-09-21:
Just to ensure nothing happens to it, here's my `Dockerfile` so far:
```
FROM ubuntu:focal

ARG SERVER=https://github.com
ARG REPO=rdeits/meshcat
ARG SHA=06c6342adc4e7e69caa88125ba6a924a1507f9e5

ENV DEBIAN_FRONTEND=noninteractive

RUN apt-get -y update

RUN apt-get -y install --no-install-recommends git npm webpack yarnpkg

RUN npm install -g license-checker

RUN git clone ${SERVER}/${REPO} && \
    cd "/$(basename ${REPO})" && \
    git checkout ${SHA}

RUN cd "/$(basename ${REPO})" && \
    yarnpkg

RUN cd "/$(basename ${REPO})" && \
    license-checker --json | \
    grep '"licenseFile":' | \
    awk '{print $2}' | \
    sed -e 's/^"//' -e 's/"$//' | \
    xargs tar --append -f /meshcat-licenses.tar
```

jwnimmer-tri, 2021-09-21:
I'm advocating for the approach of:

(1) solve this by having Drake redistribute license file(s) that come from upstream's repository, bringing it in line with how we handle almost all of our other dependencies;

(2) to the extent that the license file(s) in upstream's repository are incomplete, work with upstream to fix them;

(3) try to use existing ecosystem tools for this job.

Finishing part (1) would be sufficient to unblock pip drake's initial release.

It looks to me like your Dockerfile already meets (3)?  In any case, upstream might be able to confirm / offer advice on the tactics, and I would defer to them.

mwoehlke-kitware, 2021-09-23:
See https://github.com/rdeits/meshcat/issues/103.

closed 2021-10-08 by RussTedrake