RobotLocomotion/drake #15765: MultibodyPlant should warn when initialized in contact / beyond joint limit

edrumwri, 2021-09-11:
It's not hard to mistakenly initialize the positions of an MBP so that joint limits are violated significantly or bodies interpenetrate significantly. When simulating from such a configuration, bodies in the MBP tend to "explode".

It would be good if MBP were to warn when bodies are in contact or joints are positioned at / beyond joint limits initially. I suggest adding an initialization event to MBP to check for this.

ggould-tri, 2021-09-13:
Assigning to @sherm1 for delegation.

FYI:  The static equilibrium point of nearly every MBP joint or free body is in penetration, so unless this PR is designed very carefully it will be a warning to initialize an MBP in equilibrium.  This is the reason that we didn't pursue this in the dish station initial conditions work.

xuchenhan-tri, 2022-04-26:
@amcastro-tri I'd imagine SAP would help with the "explosion issue" even if some constraints are initially violated. Thoughts?

amcastro-tri, 2022-04-28:
I think having a warning for joint limits being violated would be a great addition, specially because we don't have a good way to visualize this problem. A contact configuration can very well be diagnosed using our visualization tools and as @ggould-tri mentions above, it is very common to start a sim from a static equilibrium configuration.

So as a first step I propose we do warn about joint limits being violated only.

Re SAP. Yes, SAP will most often recover from this, at the cost of leading to constraint forces that might not be what you want. It might be, yes, that you don't care about an abrupt transient of the solver fighting an awful initial condition. I think barking about joint limits at least once with a warning is a healthy thing to do.

xuchenhan-tri, 2022-04-28:
I think the goal here is to provide some hint to help diagnose the issue if the user accidentally set *unintended* initial condition. And the criterion to emit a warning message should be:
1. The initialization is likely unintended, and
2. there's no good way to reveal the unintended setup.

Geometries in contact are not necessarily unintended as explained above. Also, unintended contact can be easily visualized. OTOH, I don't quite see why it makes sense to initialize a joint beyond its limit (e.g. I'm not familiar with the dish station setup), and there's no easy way to reveal the mistake if an initialization beyond joint limit is indeed unintended. So I'm in favor of the warning message for the joint limit violation that @amcastro-tri proposed above.

jwnimmer-tri, 2022-04-28:
My own 2c -- I have a strong personal loathing for console spam.  The warning falsely triggering in cases where I _intend_ to exceed the limit (maybe only by a little bit?) would drive me mad.  Or, if it's _always_ a problem to exceed the limit, even by epsilon, than that is a _hard error_ (exception), not a warning.

I wonder if adding visualization for joint-limit-exceeded would be a better resolution -- it seems like that would also help in the case of simulating motion where we accidentally push up to the limit.  If I recall correctly, the limit is enforce via some kind of penalty, so if the user approaches in during motion, they would dynamics happening that cannot be explained visually.  That might apply equally well to both initial conditions or ongoing simulation?  Or have I misunderstood somewhere?

ggould-tri, 2022-04-28:
I'm not sure how to make this any clearer -- it is reasonable and often necessary to initialize joints in static equilibrium!

Consider e.g. a dishwasher door which opens downward, with joint limits at 0 (vertical) and pi/2 (horizontal) corresponding to the metal tab stops on the physical hinge.  In its open equilibrium it is is violating the pi/2 joint limit -- it has to be, since only the limit force can be balancing gravity.  In the case of our dishwasher the same is true of its closed equilbrium:  The latching force holds it against the 0 joint limit.  This system lacks a stable position that is not outside of a limit.

In static equilibrium, with the exception of objects welded, hanging straight down, or in metastable postures, every object is supported by either a collision or a joint limit.  Limit forces are a normal and necessary part of operation, and are only present when the limit is violated.  Many joints will spend most of their lives outside of their limits.

Perhaps what you actually want is to inset the action of joint limit forces from the limit itself, or to warn against positions that violate only by a certain penetration margin?

jwnimmer-tri, 2022-04-28:
For my part at least, that longer explanation helps a lot, thanks!  (And so, my hypothetical _"always a problem to exceed the limit"_ is clearly not valid.)

It does seem like we must not warn unless some kind of initial-penetration or initial-energy threshold is exceeded.

xuchenhan-tri, 2022-04-28:
@ggould-tri that explanation helped me too. Thanks. Clearly I don't have enough experience working with joint limits in actual simulations.

> If I recall correctly, the limit is enforce via some kind of penalty ...

I believe with SAP landing this will be enforced as a (soft) constraint.

I like the concept of somehow visualizing a constraint in general (not just the contact constraint where we can simply look at the geometries), but I don't have a clear idea what that would look like. @jwnimmer-tri can you elaborate?

jwnimmer-tri, 2022-04-28:
> I don't have a clear idea what [visualizing a constraint] would look like... can you elaborate?

I was imagining that the soft constraint would end up influencing the velocities somehow, and thus we could display a force vector (somewhere inbetween the two bodies that the joint connects?) that represents something roughly equivalent in world frame.  That's well into realm of what others would be better at specifying, though.

xuchenhan-tri, 2022-04-28:
I like that idea. And it could also be useful beyond just at initialization. @amcastro-tri WDYT?

jwnimmer-tri, 2022-05-04:
> In any case, a one time warning for constraints outside limits could be useful?

As I understand Grant's premise, something like 95% of simulations would print that warning, even though maybe only 1% of those have incorrectly posed their initial conditions.  That seems like too high of a false positive warning rate to be plausible.