RobotLocomotion/drake #15776: Eliminate heap allocations in MBP's GetPositions(ModelInstanceIndex), and related methods.

edrumwri, 2021-09-14:
It's presently very challenging to get positions or velocities for a single model instance without a heap allocation. Per slack discussion, a way to solve this problem would be to add output arguments to the following functions:

```
GetPositions(ModelInstanceIndex)
GetVelocities(ModelInstanceIndex)
GetPositionsAndVelocities(ModelInstanceIndex)

```
For API parity, we might want to add output arguments to:

```
GetPositionsFromArray()
GetVelocitiesFromArray()
GetActuationFromArray()
```

Output arguments would carry the side benefit of allowing one to get positions or velocities for a single model instance without needing to reserve VectorX space in user-level code by evaluating the state output port for that model instance (assuming that the appropriate evaluation function in MBP is updated to use the version of GetPositionsAndVelocities() that accepts an output argument- see https://github.com/RobotLocomotion/drake/blob/611a2f63796f5afa5cddccab7bf59279234f51aa/multibody/plant/multibody_plant.cc#L3197-L3198).

sherm1, 2021-09-14:
We'd be delighted to review a PR for this upgrade! I'd suggest marking them "(Advanced)" since most users should prefer the simpler signatures unless they are really worried about heap use.

jwnimmer-tri, 2021-11-12:
To the best of my understanding, #15900 has resolved this.  (Thanks!)  Please reply if not, and we can re-open.

closed 2021-11-12 by jwnimmer-tri