RobotLocomotion/drake #15781: atlas example prints ignoring collision group warnings

RussTedrake, 2021-09-16:
Mentioned in:
https://stackoverflow.com/questions/69199764/atlas-example-warning-in-drake-toolkit

Given @SeanCurtis-TRI 's answer "In an URDF you can specify the "group" attribute in the <link> tag. Drake doesn't make use of this attribute and is letting you know it's not being used," I think we need to update the urdf/sdf files in that example.

SeanCurtis-TRI, 2021-09-16:
Two thoughts:

1. The atlas file loaded by the example has documentation indicating that it's generated via xacro and shouldn't be edited by hand (see https://github.com/RobotLocomotion/drake/blob/master/examples/atlas/urdf/atlas_convex_hull.urdf#L2-L5). Drake doesn't include the xacro file, nor the means to re-generate this file. How do we want to address this?
2. The second half of the problem in the stackoverflow question is the quality of the error message. I'd propose that at least part of the resolution of this issue is:
    - Modification of the error message so it's more meaningful. Something along the lines of, "At least one of your URDF files has specified the "group" attribute on the <collision> tag. Drake doesn't make use of that attribute and it will be ignored."  (Note: In the stackoverflow, I'd *mistakenly* said the attribute was on the <link> tag.)
    - Change the message so it's a one-time warning (and not a per instance warning). So, it will only be fired off a maximum of one time per session.

SeanCurtis-TRI, 2021-09-16:
I'm going to submit a quick PR for the second point. If we come up with a quick resolution for the first item, I'll roll it into the same PR.

jwnimmer-tri, 2021-09-16:
> The atlas file loaded by the example has documentation indicating that it's generated via xacro and shouldn't be edited by hand... How do we want to address this?

The comment is lies; you could either ignore it, or remove it.

SeanCurtis-TRI, 2021-09-16:
So, as I look at the warnings we issue, I've got some clarifying comments/questions about some of the things being addressed.

Tags/attributes that Drake made up in the past and now warns about:

 - `link/collision@group` (appearing in the issue description) and its friend `link/visual@group` do *not* appear to be part of the [URDF spec](http://wiki.ros.org/urdf/XML/link), where the only attribute defined is "name".  This attribute was first added to the (now deleted) drake URDF schema in 1490800cb371457f793f180d1e6f1188cc6815e4. These tags appear throughout atlas files and in the pr2 file.
 - Similarly, we have `joint/dynamics@coulomb_window` that likewise is not part of [the spec](https://wiki.ros.org/urdf/XML/joint). It was added to drake's URDF schema in 1490800cb371457f793f180d1e6f1188cc6815e4. This tag only appears in [`PendulumWithFriction.urdf`](https://github.com/RobotLocomotion/drake/blob/master/examples/pendulum/test/PendulumWithFriction.urdf#L63).
 - `link/collision/drake_compliance` and its child tags `link/collision/drake_compliance/static_friction`, `link/collision/drake_compliance/dynamic_friction`, `link/collision/drake_compliance/youngs_modulus`, and `link/collision/drake_compliance/dissipation` are obsolete. The [only URDF](https://github.com/RobotLocomotion/drake/blob/master/multibody/parsing/test/links_with_visuals_and_collisions.urdf#L28-L40) in which they can be found is for [a test that doesn't care about the old tags](https://github.com/RobotLocomotion/drake/blob/master/multibody/parsing/test/common_parser_test.cc#L49). The old tags, and the parsing logic, are explicitly tested in [a different test entirely](https://github.com/RobotLocomotion/drake/blob/master/multibody/parsing/test/detail_urdf_geometry_test.cc#L635-L670).

Attributes that are valid URDF but not supported by Drake:

- Drake only supports SimpleTransmission for `transmission/type`.

For the first set, I'm converting them to log-once errors and removing all of the anachronistic usages from Drake-owned urdfs.

SeanCurtis-TRI, 2021-09-17:
I've updated the PR to include updating of all Drake-owned URDF so that these warning-inducing XML things don't induce warnings.

PR #15783 (2021-09-21): Clean up URDF parsing of unused custom tags and uses in Drake-owned URDF files
Historically, Drake had introduced a number of custom tags and attributes in URDF files. Some/many of these have long since been mothballed in practice. However, they remained in numerous URDF files owned/tracked in the repo and the parsing of those custom elements would lead to (not overly helpful) spew into the console.

This makes two changes:

  - The Drake-owned URDF files simply don't include those legacy tags/attributes anymore.
  - The warnings no longer spew on a per-instance basis. Now they do a log once.
 [... cut]
closed 2021-09-21 by jwnimmer-tri