RobotLocomotion/drake #15782: atlas example: discrete update solver failed to converge at simulation time =  0.503

RussTedrake, 2021-09-16:
As reported at
https://stackoverflow.com/questions/69199764/atlas-example-warning-in-drake-toolkit

I confirm that on master, running `bazel run //examples/atlas:atlas_run_dynamics` causes the discrete update solver to fail.  We don't catch this in CI, since CI runs the dynamics for only a fraction of a second to keep the test small.

jwnimmer-tri, 2021-09-27:
I have changed the default timestep for the atlas example so that the error no longer occurs.

@amcastro-tri wanted to investigate further, so I'm leaving this issue open for him.

amcastro-tri, 2021-09-27:
From #15814, @jwnimmer-tri run a bisect and found that #15013 (adds AVX infrastructure) apparently introduced enough round-off errors to make this Atlas test fail.

From @jwnimmer-tri:
A few notes: as a starting point, as of 0ede8b0 it was "working" in that Atlas mostly fell through the floor, but did not throw. As of a03389e, it's staying above the floor and did not throw.

In any case, the bisect eventually points to #15013 as the commit where the fault started occurring.

```
git bisect run sh -c '(bazel run --incompatible_restrict_string_escapes=false //examples/atlas:atlas_run_dynamics || false)'
...
91dd11e6eb75aa91af51b454a41b02e348af30e0 is the first bad commit
commit 91dd11e6eb75aa91af51b454a41b02e348af30e0
Author: Michael Sherman <sherm1@gmail.com>
Date:   Wed May 12 16:36:41 2021 -0700

    Add AVX infrastructure and one AVX method. (#15013)
```

amcastro-tri, 2022-02-02:
If I scroll the video posted by @jwnimmer-tri in #15814 to 0.01 secs, I see some spurious contact forces between the legs and the hip. I believe those forces are responsible for the legs being pushed sideways in that video. They might actually also be responsible for the need to use a smaller time step.

Somehow these contacts do not show with current master, and the problem is no longer observed. I am able to increase the time step back to 1.0e-3. In addition, PR #16488 adds filters to explicitly avoid those collisions.

Is there a way the bisection we did here pointed us to the wrong place @jwnimmer-tri? Maybe changes in our collision filtering were actually the cause?

jwnimmer-tri, 2022-02-02:
I don't know, but on the plus side, the failure is easy to observe (the program crashes) so you should be able to re-run it at any revision you like and see if it passes or not.

amcastro-tri, 2022-02-02:
Thanks @jwnimmer-tri, I'll give that a try.

xuchenhan-tri, 2022-04-26:
I ran this example on master bdd2b7d and had no issue even with time step of 5e-3s. (The default now is 5e-4s). Is this issue fixed? @RussTedrake @amcastro-tri

jwnimmer-tri, 2022-04-26:
A while ago (#15814), I already fixed the crash:

> I have changed the default timestep for the atlas example so that the error no longer occurs.

The remaining work on this issue is whatever @amcastro-tri wanted to do to understand more thoroughly:

> @amcastro-tri wanted to investigate further, so I'm leaving this issue open for him.

The intended follow-up was hinted at during #15814 discussions, but I'm not exactly sure what we're looking for here?  What is the "victory condition" for this ticket?

As it stands, this seems like a "personal TODO list being stored in the project's GitHub" type of issue, because this ticket is not actionable for anyone other than @amcastro-tri.  I would prefer if all Drake issues had sufficient information such that any number of project collaborates could make progress on any given issue.

In my opinion, we should ask @amcastro-tri to write up the list of tasks or questions that he had in mind for this ticket, or else we should close this as "sufficiently resolved".

xuchenhan-tri, 2022-04-26:
Like I mentioned above, the example now runs without going unstable with dt=5e-3s. That's 10x the default value set in #15814 and 5x the time step that was originally troublesome. So there must have been some more than superficial fix in the meantime.

jwnimmer-tri, 2022-04-26:
Ah, it did not click for me that you were saying you changed the timestep.  I see now.

So we have:
- Original problem report: 1e-3 at old revision => crash
- My fix: 5e-4 and middle revision => no crash
- Today: 5e-3 at tip => no crash

xuchenhan-tri, 2022-04-26:
To my surprise, `git bisect` shows that the PR that improved the stability was introduced in #16035 (i.e. prior to this PR, dt=5e-3 fails). @amcastro-tri Do you understand why this is the case? Does this indicate some lower level issues that we should be concerned about?

amcastro-tri, 2022-04-26:
The only think I can think of is exactly what @sherm1 mentions on roundoff. TAMSI is not guaranteed to converge, so it could happen that small changes in the numerics lead to a state that TAMSI could not recover from. 

In any case, IMO I think what we should do is to at least update to dt=1e-3 as in the old revision and close this issue.

xuchenhan-tri, 2022-04-27:
I'm a little iffy about just dismissing this as round-off. But if @amcastro-tri @sherm1 think there's nothing to worry about here, I'll open a PR to revert the default dt and close the issue.

xuchenhan-tri, 2022-04-27:
I can't think of anything off the top of my head. Comparing more carefully in the atlas example across the two commits may reveal something. I can take a closer look when I get more time.

amcastro-tri, 2022-05-05:
This will only get better as we merge better solvers. Not sure if worth the time to fix it @xuchenhan-tri

xuchenhan-tri, 2022-05-11:
@amcastro-tri I won't spend the time if you believe the only explanation for the instability is numerical issues. Thanks!

closed 2022-05-11 by xuchenhan-tri