RobotLocomotion/drake #15796: Roadmap for Hydroelastic Contact debut

SeanCurtis-TRI, 2021-09-21:
## Summary

Hydroelastic contact has been in development limbo for a number of years. It has intentionally been developed in a quasi sub rosa manner to allow developer freedom as we evaluated how best to implement it and integrate it with Drake. It's core functionality has been in place long enough and we've had enough chances to kick its tires and exercise its APIs that we can see the end of the tunnel. We're *just about* ready to promote it to a full member of the Drake feature set.

There are still a number of items that we want to achieve before we pull the trigger. As promotion to "full feature" entails all of the cost of API changes (i.e., deprecation) for any future changes. Below we've listed various tasks related to the continuing development of hydroelastic contact. They've been coarsely categorized into prerequisites for the debut, and those that can follow after.

### Prerequisites for promotion to full feature
  - [ ] Technical debt clean up
      - [ ] Eliminate the `{Volume|Surface}Vertex` types.
          - These are thin, impoverished wrappers of Vector3<T>. We'll simply expose the functional type.
      - [ ] Eliminate `Volume*Index` and `Surface*Index`.
          - The index types introduce a fair amount of verbosity without much value. The public API doesn't have functions that take multiple index types, therefore the primary benefit is missing. On the other hand, constructing meshes is a penance.
      - [ ] Clean up documentation to remove references to "soft" geometry.
          - It has been decided that the term "soft' will remain exclusively in the domain of truly deformable geometry. In the hydroelastic domain we'll use the term "compliant" to contrast with "rigid".
      - [ ] Address the "elastic_modulus" material lie.
          - Hydroelastic contact makes use of a property called "elastic modulus". While that would typically be a material property, the way hydroelastic property uses it, it *isn't* actually a material property. This will inevitably lead to surprise and confusion. We've brainstormed some ways to alleviate this:
              - Simply rename it to something proprietary (e.g., "hydroelastic modulus") and document its idiosyncracies.
              - Change to a *different* value that has a stronger "material"-like property. Currently, the use of "elastic modulus" is essentially "maximum pressure" (i.e., the most pressure that a geometry can exert at contact at its deepest penetration). If, instead, we were to change it to "pressure per unit penetration distance", then it would be decoupled from the scale of the geometry. (Note: there's some subtle nuance here when working with volume meshes with anisotropic measures and a single layer of tets.)
  - [ ] Features
      - [ ] User guide
          - A doxygen-rendered guide which describes (at an intuitive level) hydroelastic contact, with appropriate references to published papers and code. Its emphasis is to be a one stop guide for users to know how to enable hydroleastic contact, the vagaries of working with it, how to define appropriate properties, and various tips and tricks culled from the tribal knowledge.
      - [ ] Blog post
          - This isn't a "prerequisite" so much as it is coincident. Some of the text will probably be borrowed from the user guide.
      - [ ] Full support for a "polygonal" contact patch.
          - The polygonal contact patch is a representation of the contact surface with fewer total elements.
          - We will also have full visualization support in `drake_visualizer` (but not `Meshcat`, see below).
      - [ ] Confirm policy for coordinating contact surface type (sparse polygon or dense triangle) with `MultibodyPlant` discrete/continuous mode.
          - The polygonal mode is currently the *natural* partner of `MulitbodyPlant`'s efficient discrete mode. But in master (as of 9/20/2021), the user must explicitly opt in. Potentially, we would want to change that default. Or not. Either way, it needs to be explicitly considered.
      - [ ] Tutorial (in python notebook)
          - Allow users to play with hydroelastic properties interactively and immediately see the effect. A powerful complement to the user guide.
      - [ ] All relevant public APIs fully bound in python (i.e., normal exercise of hydroelastic contact should be equally available in python).

### Post-promotion development
  - [ ] Visualization of hydroelastic contact results in Meshcat (C++ version, *not* python)
  - [ ] Support compliant-compliant contact.
     - Remember, the current state only supports compliant-rigid contact.
  - [ ] Support compliant geometries from arbitrary meshes (instead of just from platonic primitives).
     - There are two different complementary avenues to support this
         - Define an off-line workflow that users can use to take an OBJ and create a volume mesh and pressure field on disk and have Drake parse those files.
         - Implement, in Drake, that pipeline on-line.
  - [ ] Extend examples to include a case where hydroelastic properties are fully defined in URDF/SDF files.
  - [ ] Allow configuring `MultibodyPlant` to use a lower order of integration  (and the polygonal contact surface) even in continuous mode.

rpoyner-tri, 2021-09-21:
There's a lot of soft/compliant term use to review:

```
$ git grep -i '\bsoft\b' |grep -v /dev/ |wc -l
299
```

DamrongGuoy, 2021-11-01:
I confirm that for `Technical debt clean up`, we are quite good now, right? All the data structures (*Vertex, *Index, *Mesh, *Field\*) are quite stable now. It's safe to expect no major merge conflict in my draft PR #15545 (compliant-compliant contact) anymore, right?  I'll focus this month on compliant-compliant contact.

SeanCurtis-TRI, 2021-11-02:
@DamrongGuoy I don't know how entangled the compliant-compliant contact is with the compliant-rigid mesh contact, but that code is getting heavily touched due to incorporating the `PolygonSurfaceMesh`. Given the size of that effort, I would prefer not to have to rebase it to accommodate compliant-compliant. Keep that in mind.

DamrongGuoy, 2022-01-05:
@joemasterjohn will take care of this roadmap now, so I can focus on compliant-compliant contact #16040.

I remove myself from the list of assignees.

rpoyner-tri, 2022-04-14:
There is not much value left in keeping this open. I will peel off the remaining items of interest to separate issues, and close this one.

jwnimmer-tri, 2023-07-30:
@rpoyner-tri is the peeling complete?  Can we close this now?

rpoyner-tri, 2023-08-29:
I've gone through the original checklist. I'm pretty satisfied we have at least MVP support for everything. The only open item is post-release performance tuning; I feel like important pain points will make themselves obvious over time.

Closing.

closed 2023-08-29 by rpoyner-tri