RobotLocomotion/drake #15800: Use lcm-logplayer-gui with MeshCat.

DamrongGuoy, 2021-09-21:
Please let me know if this issue is really not an issue.  I'm not familiar with MeshCat, so I might misunderstand.

### History

I am using `//tools:drake_visualizer` with `lcm-logplayer-gui` like this:
```
bazel run //tools:drake_visualizer &
bazel build @lcm//:lcm-logplayer-gui
bazel-bin/external/lcm/lcm-logplayer-gui  <my_lcm-log_file_from_my_simulation> &
```
It is very convenient to playback LCM messages without actually running the simulation again. The `lcm-logplayer-gui` allows me to play slower, faster, or to pause at a certain time step. I can adjust visualization settings and replay again.

### Request

Ideally I would like `MeshCat` to listen to LCM messages from `lcm-logplayer-gui` too. However, I will accept any solution that allows me to do similar things.

RussTedrake, 2021-09-21:
I see.  Meshcat doesn't use lcm, so lcm logs aren't going to do the job.

Instead, we can record (and playback) animations.  Here is an example:
https://people.csail.mit.edu/russt/uploads/compass_gait.html

Click "Open Controls" and you will see play/pause/reset and a time slider all available for you.  Implementing the SetAnimation method in the c++ Meshcat is the next (and one of the last) main features I need to reimplement in c++.  I'm happy to have an issue to track it.

SeanCurtis-TRI, 2021-09-22:
@RussTedrake This (recording animations) presumably works fine for the poses of frames, but as we add support for visualizing contact surfaces and the like, I presume we're going to need a different mechanism, right?

I believe that's something that @DamrongGuoy's lcm-based workflow gets him for free.

All of this would probably be resolved if we had (in Drake) the ability to simply save entire contexts and then play them back. But I believe that's still a large, distant endeavor.

RussTedrake, 2021-09-26:
Just to be clear -- if someone wants to visualize the results of a lcm log file (e.g. of estimated robot state) in meshcat instead of drake visualizer, there is no impediments to that.  In both cases, you would use lcm-logplayer-gui to play the state messages, and have a small drake app that subscribes to that lcm and publishes to meshcat.

The *only* thing that is different here is if you were relying on some lcm viewer draw message in your log files reproducing the visualization directly to drake visualizer; that part won't work with meshcat.  

(I had a hallway conversation with someone a few days ago that made me realize this should be clarified).

I've updated the issue title to reflect my intention to only resolve this, right now, by implementing SetAnimation.  @DamrongGuoy -- if you feel that this does not capture the intention of your issue, then feel free to rename it again, but then you'll need assign it to someone else.  ;-)

commit d709517d (2021-10-08): [geometry:meshcat] Adds SetAnimation (#15876)

* [geometry:meshcat] Adds SetAnimation

Provides a MeshcatAnimation class and Meshcat::SetAnimation.

Includes an upgrade to my test script which prints more narrow diffs;
this proved invaluable for debugging this complicated message type.

Resolves #15800.