RobotLocomotion/drake #15802: Build errors with XCode 13 (clang-1300.0.29.3)

soonhokong, 2021-09-22:
FYI, this week, Apple released [XCode 13](https://developer.apple.com/documentation/xcode-release-notes/xcode-13-release-notes) which includes `clang-1300.0.29.3` and it generates build errors when I compile Drake with it:

Some errors:
```
Use --sandbox_debug to see verbose messages from the sandbox
geometry/optimization/iris.cc:326:21: error: loop variable '[geomA, geomB]' creates a copy from type 'const std::pair<drake::geometry::GeometryId, drake::geometry::GeometryId>' [-Werror,-Wrange-loop-construct]
    for (const auto [geomA, geomB] : pairs) {
                    ^
geometry/optimization/iris.cc:326:10: note: use reference type 'const std::pair<drake::geometry::GeometryId, drake::geometry::GeometryId> &' to prevent copying
    for (const auto [geomA, geomB] : pairs) {
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
                    &
```

```
Use --sandbox_debug to see verbose messages from the sandbox
solvers/choose_best_solver.cc:295:19: error: loop variable 'solver_id' creates a copy from type 'const drake::solvers::SolverId' [-Werror,-Wrange-loop-construct]
  for (const auto solver_id : solver_ids) {
                  ^
solvers/choose_best_solver.cc:295:8: note: use reference type 'const drake::solvers::SolverId &' to prevent copying
  for (const auto solver_id : solver_ids) {
       ^~~~~~~~~~~~~~~~~~~~~~
                  &
```

SeanCurtis-TRI, 2021-09-22:
Oops. @jamiesnape had done a pass through some time ago to clean up this kind of thing. This is a recent addition. Well, it's easy enough to resolve. Presumably, this is the only such error that has crept back in, yes?

soonhokong, 2021-09-22:
> Presumably, this is the only such error that has crept back in, yes?

I am not sure. I think by the time Drake supports XCode 13, the team needs to update the CI and run a complete check. Until then, this is not a high-priority item?

/cc @BetsyMcPhail

SeanCurtis-TRI, 2021-09-22:
@rpoyner-tri as the man who is basically wearing @soonhokong's mac hat, I'm assigning this to you. Feel free to pass it along as appropriate.

rpoyner-tri, 2021-09-23:
So. XCode 13 requires a newer macOS than our test box `macsim` currently has. Is there any reason not to upgrade it, and let CI machines hold down whatever the oldest-supported configuration is?

jwnimmer-tri, 2021-09-23:
As long as you stay on "Big Sur" (don't go to "Monterey"), I think we'll be OK.  It means that we won't be able to repro CI failures that _only_ happen with XCode 12 (that don't happen on 13), but we should upgrade CI to XCode 13 as soon as you have it working.

rpoyner-tri, 2021-09-23:
`macsim` is now at macOS 11.6 "Big Sur". I'll see about the XCode upgrade.

rpoyner-tri, 2021-09-23:
Xcode is upgraded. Working to repro now.

rpoyner-tri, 2021-09-23:
Warnings about inadvertent copies like those listed above are easy to reproduce and fix. However, there seems to also be some change of behavior in the TypeHasher.

```
[----------] 1 test from TypeHashTest                                                                                                                                                                   
[ RUN      ] TypeHashTest.WellKnownValues                                                                                                                                                               
common/test/value_test.cc:406: Failure                                                                                                                                                                  
Expected equality of these values:                                                                                                                                                                      
  internal::TypeHash<T>::value                                                                                                                                                                          
    Which is: 5139326514624727031                                                                                                                                                                       
  size_t(hasher)                                                                                                                                                                                        
    Which is: 10380075647220874350                                                                                                                                                                      
  for name                                                                                                                                                                                              
    Which is: std::__1::shared_ptr<double>                                                                                                                                                              
  for __PRETTY_FUNCTION__                                                                                                                                                                               
    Which is: void drake::test::CheckHash(const std::string &) [T = std::shared_ptr<double>]                                                                                                            
common/test/value_test.cc:406: Failure                                                                                                                                                                  
Expected equality of these values:                                                                                                                                                                      
  internal::TypeHash<T>::value                                                                                                                                                                          
    Which is: 18366249326615519886                                                                                                                                                                      
  size_t(hasher)                                                                                                                                                                                        
    Which is: 6434182069455397803                                                                                                                                                                       
  for name                                                                                                                                                                                              
    Which is: std::__1::pair<int,double>                                                                                                                                                                
  for __PRETTY_FUNCTION__                                                                                                                                                                               
    Which is: void drake::test::CheckHash(const std::string &) [T = std::pair<int, double>]                                                                                                             
common/test/value_test.cc:406: Failure                                                                                                                                                                  
Expected equality of these values:                                                                                                                                                                      
  internal::TypeHash<T>::value                                                                                                                                                                          
    Which is: 6095216155716852889                                                                                                                                                                       
[... cut ...]
  for __PRETTY_FUNCTION__                                                                                                                                                                               
    Which is: void drake::test::CheckHash(const std::string &) [T = std::vector<std::pair<const double, std::shared_ptr<Eigen::Matrix<double, 3, 3, 0>>>>]                                              
[  FAILED  ] TypeHashTest.WellKnownValues (0 ms)                                                                   
```

jwnimmer-tri, 2021-09-23:
It seems like Apple's PRETTY_FUNCTION may have stopped adding the `__1` noise into the typenames?

Thus, this line from the unit test is probably wrong now, and needs to also check the Apple Clang version:
```c++
const std::string stdcc = kApple ? "std::__1" : "std";
```

rpoyner-tri, 2021-09-23:
Yes, some patch to the kApple magic fixes it; not sure I have the right formulation yet. Found that just before returning here to see your message.

Next bit is some failure in render_engine_vtk_test.cc. Passes cleanly in `-c dbg` build, but fails reliably in release build.

```
[ RUN      ] RenderEngineVtkTest.RemoveVisual
geometry/render/test/render_engine_vtk_test.cc:321: Failure
Expected equality of these values:
  label.at(x, y)[0]
    Which is: 32764
  expected_outlier_label_
    Which is: 0
Label at: (10, 10) for test: First sphere added in remove test
geometry/render/test/render_engine_vtk_test.cc:321: Failure
Expected equality of these values:
  label.at(x, y)[0]
    Which is: 32764
  expected_outlier_label_
    Which is: 0
Label at: (10, 469) for test: First sphere added in remove test
```
... and many similar ones ...  Any thoughts, @SeanCurtis-TRI ?

rpoyner-tri, 2021-09-23:
The RenderEngineVtkTest mystery deepens. I think I may be looking like a wild write of some sort. As I add and remove print statements from the code, the locus of failure moves around also.

jwnimmer-tri, 2021-09-23:
Did it pass with XCode 12?  I've _never_ gotten the Vtk tests to pass when run over ssh to macsim.

rpoyner-tri, 2021-09-24:
I can't answer the Xcode 12 question until I figure out how to install multiple versions and see if I can get `xcode-select` to work.

As I fiddle around more, i wonder if I am staring at an optimizer bug. `-c dbg` always works, and release build works if I insert a statement that *reads* a specific member variable at a small number of auspicious places. The symptom is that variable (that explicit code always sets to some non-zero value) getting set to 0, unless I insert the magic read statement.

jwnimmer-tri, 2021-09-27:
Weird test errors aside, it's probably worth submitting at least the _compiler error_ fixes to master sooner rather than later, so that our end users are unblocked.

rpoyner-tri, 2021-09-28:
Back to render_engine_vtk_test: confirmed that it passes in release builds with injection of either `--copt=-O0` or `--copt=-O1`, but fails with `--copt=-O2`, or with release-mode defaults.

rpoyner-tri, 2021-09-29:
As of the three patches merged above, drake builds and passes tests with Apple clang 13.

closed 2021-09-29 by rpoyner-tri