RobotLocomotion/drake #15810: Feature Request: SetAcrobotParameters method in AcrobotPlant Class

vyas-shubham, 2021-09-24:
In the `AcrobotPlant` class, there is a function to set the MIT acrobot parameters: `SetMITAcrobotParameters()`. However, it would be useful to have a method to set parameters for other acrobots as well. This is useful for other labs working with an acrobot system and would like to use Drake for trajectory optimization. 

This plant is different form the one from urdf or the `MultibodyPlant` created using `MakeAcrobotPlant()` function as `MultibodyPlant` does not (yet) include damping when performing trajectory optimization. The `AcrobotPlant` class has damping implementation which is useful for performing better trajectory optimization.

Hence, to perform trajectory optimization and stabilization for other acrobot setups, the `SetAcrobotParams()` class would be useful. There exists the `acrobot_params_named_vector.yaml` file to change the default parameters, but compiling drake from source to use this method of changing parameters means one has to give up the SNOPT solver bundled with the binary installation of drake. Having SNOPT is quite useful for the trajectory optimization.

jwnimmer-tri, 2021-09-24:
The AcrobotPlant offers this function:

```c++
class AcrobotPlant {
  // ...
  const AcrobotParams<T>& get_mutable_parameters(Context<T>* context) const;
};
```

There _is_ a bug here, in that the return type should not be `const`.  Assuming we fix that, would it be sufficient for your needs?

```c++
AcrobotPlant plant;
auto context = plant.CreateDefaultContext();
auto& params = plant.get_mutable_parameters(context.get());
params.m1 = 0.5;
params.m2 = 0.6;
// ... etc ...
```

closed 2021-09-24 by jwnimmer-tri

vyas-shubham, 2021-09-27:
This works indeed. Thank you.

PR #15812 (2021-09-24): [acrobot] Remove spurious const
There is already a pydrake unit test for this, it's just that pybind11 always discards "const" qualifiers so we never noticed.

Closes #15810.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15812)