RobotLocomotion/drake #15815: CMake option to use user-provided spdlog (and thus, fmt)

jwnimmer-tri, 2021-09-24:
Some Drake users would like to build Drake from source using CMake, and to supply their own already-compiled versions of `spdlog` and `fmt`, instead of using the Ubuntu 20.04 revisions of those libraries, so that they can use some newer features not available in Ubuntu.

As background -- by default, drake uses the Ubuntu 20.04 build of these two libraries, so that is has binary (ABI) compatibility with other software compiled for Ubuntu 20.04.  Our binary releases will continue to do so.

Here, the request is to allow users to opt-of of those versions, and instead supply their own (newer) versions of `spdlog` and `fmt`.

jwnimmer-tri, 2021-09-24:
I think the smart thing here would be to address this question for _all_ of the dependencies used by Drake's public headers.

Taking the list from https://github.com/RobotLocomotion/drake/issues/7451#issuecomment-926851544, that means:
- Eigen
- fmt
- spdlog
- yaml-cpp

jwnimmer-tri, 2021-10-02:
> yaml-cpp

In #15868, I've proposed removing this as a public dependency.

jwnimmer-tri, 2022-02-15:
@BetsyMcPhail @mwoehlke-kitware FYI there are some users who could immediately benefit from this, so I've upgraded its priority on the project board, to be the next ticket that's On Deck.

I think the best implementation strategy here might be to change Drake's CMakeLists to:

(1) Define a CMake option for a user to opt-out of Drake's default for Eigen; by default, we keep using the same Eigen as currently (per repository.bzl); when the users opts-out, we do something like `find_package()` or whatever CMake thing makes sense to locate the Eigen that they want to use;

(2) Define a CMake option for a user to opt-out of Drake's default for fmt and spdlog (I don't think opting-out of only one of them makes sense); by default, we keep using the same spdlog and fmt as currently (per repository.bzl); ditto on `find_package()` or similar when the user opts-out.

(3) If the user chooses either of those opt-out options, then we find the already-installed library that they want to use and override Drake's WORKSPACE to refer to it by passing [--override_repository](https://docs.bazel.build/versions/main/command-line-reference.html#flag--override_repository) in the [CMake-generated rcfile](https://github.com/RobotLocomotion/drake/blob/master/cmake/bazel.rc.in) for Drake.

For example, we would pass the rcfile flag `build --override_repository=eigen=/something/eigen` where the `/something/eigen` was some build-specific temporary folder created by Drake's CMakeLists that contains the files Bazel needs to refer to Eigen, i.e.,

```console
jwnimmer@call-cps:/tmp/eigen$ ls -l
total 4
-rwxrwxr-x 1 jwnimmer jwnimmer 172 Feb 14 23:58 BUILD.bazel
lrwxrwxrwx 1 jwnimmer jwnimmer  19 Feb 14 23:57 include -> /usr/include/eigen3
-rw-rw-r-- 1 jwnimmer jwnimmer   0 Feb 14 23:55 WORKSPACE
jwnimmer@call-cps:/tmp/eigen$ cat BUILD.bazel 
cc_library(
    name = "eigen",
    hdrs = glob(["include/**"]),
    includes = ["include"],
    visibility = ["//visibility:public"],
)
```

The BUILD file contents could be hard-coded, but the `include` symlink to find the library would be based on `find_package()` or similar.

To figure out what needs to go into the `/something/foo` folder for each, run a normal Drake bazel build (e.g., `bazel build //common:*` and then inspect the contents of `bazel-drake/external/{eigen,fmt,spdlog}` for what's there.  Those files (the output of a repository rule) are the ones we need to match here.

The spdlog BUILD.bazel would be something like this:
```py
cc_library(
    name = "spdlog",
    hdrs = glob(["include/**"]),
    defines = ["SPDLOG_COMPILED_LIB", "HAVE_SPDLOG"],
    linkopts = ["-lspdlog", "-pthread"],
    deps = ["@fmt"],
    visibility = ["//visibility:public"]
)
load("@drake//tools/install:install.bzl", "install")
install(name = "install")
```

The fmt BUILD.bazel example would be best taken from a macOS sample build (where we use pkg-config); the Ubuntu 20 rule to use spdlog's embedded fmt isn't quite the right fit here.

jwnimmer-tri, 2022-02-18:
It requires to touch an empty `WORKSPACE` file, yes.  That doesn't seem like a meaningful problem?

mwoehlke-kitware, 2022-02-22:
Okay... fun things I've found:
- Drake ***only*** works with fmt 6.1.2 (possibly with 6.2.x). Newer versions (fmt ≥ 7) result in build errors in Drake.
- fmt 6.x installs its headers to `include` by default, but Drake ***requires*** that they are installed in `include/fmt`. **Use `-DFMT_INC_DIR:PATH=include/fmt`.** (fmt ≥ 7 DTRT by default but aren't usable; see previous point.)
- spdlog should be built with external fmt (`-DSPDLOG_FMT_EXTERNAL:BOOLEAN=ON`).

mwoehlke-kitware, 2022-02-22:
> How involved would it be to fix that?

I have no idea. There are static asserts thrown about no formatter being available. (I don't think I have logs any more.)

duanyutong, 2022-02-22:
To clarify, we haven't built drake with fmt 8.0.1. We use the prebuilt drake release in an image with fmt 8.0.1 installed to the default system location.

https://github.com/DexaiRobotics/drake-torch/blob/master/drake-torch.dockerfile#L165-L173

jwnimmer-tri, 2022-02-22:
That's confusing because we use `fmt==8.1.1_1` on homebrew, with no problems.

In any case, I think it's fine to merge the current PR where only the backdated versions of spdlog & fmt appear to succeed.  Then separately, we can open a new issue to follow-up to make Drake forward-compatible with the latest spdlog & fmt.

mwoehlke-kitware, 2022-02-22:
Okay, fmt=8.1.1 and spdlog=1.9.2 seems to work. I'm pretty sure I had problems with fmt 7 and fmt master.

jwnimmer-tri, 2022-02-22:
Hmm.  The `#include <fmt/ostream.h>` atop `symbolic_decompose.cc` is supposed to have the user-defined formatter come from `Expression::operator<<`.  Maybe something with the ostream compatibility API has changed in recent fmt.

mwoehlke-kitware, 2022-02-23:
As discussed on Bazel, spdlog/master is apparently ancient and crufty. So, probably not surprising it doesn't work.

It's also possible things fall apart if the fmt and spdlog versions are not matched (e.g. I'm not sure that 7.x/1.6 or 8.x/1.6 work).

jwnimmer-tri, 2022-03-01:
Upthread, I'd added "Eigen" into the requirements here, but perhaps most clear would be to keep this issue about fmt and spdlog only, and treat Eigen separately afterwards.  I've filed #16693 to that end.  Once #16652 merges, we can close out this issue.

PR #16652 (2022-03-02): Allow external fmt, spdlog
Add logic (only when using the CMake wrapper) to allow the user to select and use external version(s) of fmt and/or spdlog.

Fixes #15815.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16652)
closed 2022-03-02 by jwnimmer-tri