RobotLocomotion/drake #15828: //geometry:meshcat_visualizer_test times out under mac asan

ggould-tri, 2021-09-27:
Observed:
* https://drake-jenkins.csail.mit.edu/user/ggould-tri/my-views/view/Buildcop/job/mac-big-sur-clang-bazel-nightly-everything-address-sanitizer/228/
* https://drake-jenkins.csail.mit.edu/user/ggould-tri/my-views/view/Buildcop/job/mac-catalina-clang-bazel-nightly-address-sanitizer/712/
* https://drake-jenkins.csail.mit.edu/user/ggould-tri/my-views/view/Buildcop/job/mac-catalina-clang-bazel-nightly-everything-address-sanitizer/716/

Typical builds before were circa 10s, while the timeout is at 300s, so this is not a small regression that happened to cross an arbitrary time threshold.  This is likely due to #15788 or some close relative thereof.

This issue may already be resolved in more recent commits; if so it will be closed.  Otherwise it will be worked around by putting a no-asan tag on this test; however the underlying issue remains.

ggould-tri, 2021-09-27:
CI indicates this bug was solved this morning, possibly by #15823.  Closing now.

ggould-tri, 2021-09-28:
Reopening: This had another CI failure.

https://drake-jenkins.csail.mit.edu/view/Production/job/mac-big-sur-clang-bazel-nightly-address-sanitizer/227/

ggould-tri, 2021-09-28:
The workaround is now in place.  This issue remains open as the underlying bug presumably remains.  I am assigning it a low priority for now, since it's not blocking anything.

closed 2021-09-29 by rpoyner-tri

jwnimmer-tri, 2021-09-29:
When merging commits to master, please use descriptive subject lines ("Speed up meshcat C++ mtl parsing"); the issue number can be in the commit body, as supplemental information.  Issue numbers do not describe anything, on their own.

PR #15842 (2021-09-29): [geometry:meshcat] fix #15828
Resolves #15828

The regex that I was doing on the obj file contents was tragically
slow, especially on mac debug builds.  I have replaced it with a find,
then a more narrow regex.

Before this PR, running
```
bazel run --config=apple_debug //geometry:meshcat_visualizer_test
```
would execute in ~195 seconds.  After this PR, it executes in ~4 seconds.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15842)
commit 3cb2db0b (2021-09-29): [geometry:meshcat] fix #15828 (#15842)

* [geometry:meshcat] fix #15828

Resolves #15828

The regex that I was doing on the obj file contents was tragically
slow, especially on mac debug builds.  I have replaced it with a find,
then a more narrow regex.

Before this PR, running
```
bazel run --config=apple_debug //geometry:meshcat_visualizer_test
```
would execute in ~195 seconds.  After this PR, it executes in ~4 seconds.