RobotLocomotion/drake #15872: Upgrade to ibex-2.8.6

soonhokong, 2021-10-04:
We've found an issue with the current version of ibex (ibex-2.7.4) (see #15789). We will move to `ibex-2.8.6` to resolve the problem. For the record, `ibex>=2.8.7` dropped the CLP support which is the only freely available LP solver (up-to ibex-2.8.6). There is an attempt to bring the CLP support back to ibex, but it's still in-progress.

Action Items:
 - [ ] Prepare `.deb` for `ibex-2.8.6`.
 - [ ]  Provide `ibex@2.8.6` in https://github.com/RobotLocomotion/homebrew-director/tree/master/Formula. Note that this requires generating homebrew bottles.
 - [ ] Drake needs to use a version of dReal which is compatible with `ibex-2.8.6`.
 - [ ] Need to update `setup` scripts to use `ibex-2.8.6`.
 - [ ] `IbexSolver` implementation needs a few changes to be compatible with `ibex-2.8.6`

/cc @BetsyMcPhail for heads-up for infrastructure updates.
/cc @RussTedrake

jwnimmer-tri, 2021-10-04:
> Prepare .deb for ibex-2.8.6.

Where is the deb rebuilding process documented, or more specifically, where does the source code for that `*.deb` originate from?  Since the code is LGPL, we need to ensure that we can provide the sources anytime we provide precompiled binaries.

soonhokong, 2021-10-04:
It will be uploaded to https://launchpad.net/~dreal/+archive/ubuntu/dreal/+packages where you can download source code as well.

RussTedrake, 2021-10-20:
ftr -- i have ibex 2.8.4 building from source in bazel now.  will PR it shortly.

jwnimmer-tri, 2022-05-06:
The latest tag to dReal is 4.21.06.2.  I think we'll need a new version number tag, before we can upgrade.

mwoehlke-kitware, 2022-06-02:
Just to add a wrinkle... the currently-used-by-drake version of ibex does not compile on Jammy.

mwoehlke-kitware, 2022-06-02:
See also https://github.com/ibex-team/ibex-lib/issues/461; it appears upstream has re-added CLP support?

jwnimmer-tri, 2023-01-27:
Not relevant anymore, per #18156.

closed 2023-01-27 by jwnimmer-tri