RobotLocomotion/drake #15916: RenderRpc: Evaluate and select RPC protocol

zachfang, 2021-10-13:
Evaluate different RPC protocols and select the most promising one for the initial implementation. The decision can be revisited later on if needed.

zachfang, 2022-03-08:
The project has decided to use the [curl library](https://curl.se/libcurl/) with the HTTP protocol to communicate between client/server.

closed 2022-03-09 by svenevs

jwnimmer-tri, 2022-05-05:
Here was our thinking:

(1) Sending large files (multi-megabyte scenes) over an RPC in-band is generally frowned upon.  Not many RPC middlewares perform well with such large payloads.

(2) We presume there is just 1 client implementation (Drake), but >>1 server implementations (Blender, Godot, Unity, Unreal, Ospray, etc.).  We want to choose a protocol that will be very easy to adopt in _all_ of those rendering environments.  Even everyday things like gRPC are sometimes difficult to link into those environments, and in general many RPC middlewares will only offer 1 concrete implementation, so if it doesn't work for a certain render server, we're out of luck. On the other hand, there are a wide variety of HTTP REST server implementations available, so it should be plausible to find _one_ that works for any given renderer.

PR #16723 (2022-03-09): [geometry/dev] add HttpServiceCurl implementation
Cannot merge until

- [X] #16702 is merged.
- [X] This branch is rebased.

Closes #15916.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16723)