RobotLocomotion/drake #15919: RenderRpc: RenderEngineRpcClient implementation

zachfang, 2021-10-13:
Implementation of RenderEngineRpcClient. Besides the implementation, this ticket also includes unit tests to ensure code coverage and sufficient documentation.

kunimatsu-tri, 2022-01-13:
WIP PR submitted to #16234.

kunimatsu-tri, 2022-01-13:
Concern arose that the choice of gltf serializer might become an information bottleneck for us. We should be cautious about this point when making our final decision. For more details, refer to [the slack thread](https://drakedevelopers.slack.com/archives/C02HMU7B15Y/p1641510759008900).

zachfang, 2022-01-28:
The first pass of the client implementation is merged via #16234.

zachfang, 2022-03-08:
All the prototyping is done mainly in #16234, and the subsequent PRs (#16533, #16504, #16556, and #16609).

closed 2022-03-08 by zachfang