RobotLocomotion/drake #15931: Consider a mechanism to deprecate geometry property names.

DamrongGuoy, 2021-10-15:
I convey the analysis of this problem from @rpoyner-tri .  Thanks so much again.  In a spirit, this issue is similar to #12214.

# Problem Statement
The class [GeometryProperties](https://github.com/RobotLocomotion/drake/blob/54efe22821365359476f77b458e478551b08a1be/geometry/geometry_properties.h#L225) have some public API, and parts of that API are the string names of the properties. For example, in PR #15911, we want to respell “elastic_modulus” to “hydroelastic_modulus”. How to deprecate a property name?

# Proposal
Add the ability to name “deprecated aliases” to geometry properties. Some observations:
- A property is in a “group”, so there are two levels of maps and lookups. Do we want to support deprecation for just property names? Group names? Both? In the immediate case, we could ignore group names if that helps.
- Properties aren’t statically defined, but in fact, are created at run-time by AddProperty().
  - Properties don’t need to exist at all. If no one calls AddProperty(), they will just be missing and that is fine.
- This suggests we could add some new method, say,
  - AddDeprecatedPropertyAlias()
  - This would take the old name(s), the new name(s), and probably a deprecation date string.
  - The result would be that the old name would still work, but using it would emit a log_once style message.
  - Adding an alias will change the behavior of HasProperty() and GetProperty(), so that using the old name would have the same result as using the new name, with the added side effect of a log_once message.
- It should be the case that all of the API of property deprecation could be private and/or internal. It’s not necessary for users to have access to it.

# Implementation notes
- Add a separate map of old-name ⇒ new-name. Probably need to ensure that old names and new names are completely disjoint sets. This is hard within the code; may have to rely on code review to do that.
- Do the lookup the existing way. If it fails to find something, then check the alias map. If we get a hit, we need to substitute the new name and look up again. Also, we need to issue a message *for this particular name* and remember that we have warned for it.
- So maybe the mapping is actually an old-name ⇒  struct with stuff in it.
- Struct contains at least:
   - New name
   - Deprecation date
   - A flag we can set to remember we sent the message
- When regular lookup fails, we consult the alias map. If the message flag is clear, then we use the names and the date string to build and send a message, and then set the flag. Then we substitute the new name and re-do the lookup as before.
- Conceptually the alias mapping is global, so we could (for example) build the alias table as a singleton that various property instances all share.

# Objections
Is this more deprecation effort than we want to spend? Is it slow?

sherm1, 2021-10-15:
Since hydroelastics has always been presented as "experimental" up to now, do we really need to deprecate this? We haven't been deprecating hydroelastic API changes.

rpoyner-tri, 2026-11-11:
Reassigning to @SeanCurtis-TRI, since this is a geometry design question in your area rather than something I'm working on. Please unassign yourself or push back if that's wrong.