RobotLocomotion/drake #15943: Proposal for Context-local scratch storage sugar

sherm1, 2021-10-19:
### Background
For Drake to be thread safe, each thread must have its own mutable memory. We normally handle that with a per-thread Context that contains a cache. In many cases a thread just needs some scratch memory and doesn't need the elaborate coherency support provided by the Context Cache. It is tempting to do this with a mutable member in a LeafSystem, but that destroys thread safety. The best approach to date is to allocate a cache entry with no dependencies and a no-op computation method. That works, but is ugly.

@edrumwri started a Slack thread [here](https://drakedevelopers.slack.com/archives/C2CHRT98E/p1631722083010600) to upgrade this.

### Proposal

To `SystemBase` add
```c++
template <typename ValueType>
CacheIndex DeclareScratchStorage(std::string description, const ValueType& initial_value);
```

To `ContextBase` add
```c++
// Note that this is a const method.
template <typename ValueType>
ValueType& get_mutable_scratch_storage(CacheIndex) const;

// This is not likely to get much use, but could be useful in long calculations
// where the scratch storage has been written on earlier.
template <typename ValueType>
const ValueType& get_scratch_storage(CacheIndex) const;
```

We would note that
- The same chunk of memory is returned every time, untouched by Drake. Thus allocations and sizing can be done late.
- The contents of scratch storage must not be used to affect the trajectory of the simulation. That is, it must not be treated like state.
- The storage is actually located in the Context Cache, along with other cache entries. This is needed in case the user screws up and gets an error message from the cache infrastructure.

### Compatibility note
To cut out annoying boilerplate for regular cache entries in the same way, we should consider adding to `ContextBase`
```c++
template <typename ValueType>
const ValueType& get_cache_value(CacheIndex) const;

template <typename ValueType>
ValueType& get_mutable_cache_value(CacheIndex) const;
```
These would be synonymous with the scratch methods proposed above, but would throw based on cache value validity, while the scratch methods will never throw.

Thoughts, please.

SeanCurtis-TRI, 2021-10-19:
> ...subclassing...

Oh, yeah. That makes perfect sense.  Thanks for the follow up on that.

jwnimmer-tri, 2022-02-01:
A few months later now, we have several more examples of scratch use within Drake proper, so we should be able to evaluate the proposed API here against those call sites.

sherm1, 2022-02-11:
Per [slack discussion](https://drakedevelopers.slack.com/archives/C2CHRT98E/p1644622933266889?thread_ts=1644621112.226229&cid=C2CHRT98E) it is always a mistake to `Eval()` a scratch entry. That causes the "up to date" bit to get set so that a later attempt to get mutable access will fail. @jwnimmer-tri suggested using an AlwaysThrowCalc() rather than a NoOpCalc() for those to catch ill-advised Evals.

jwnimmer-tri, 2024-11-01:
FYI an inventory of all current uses:
* `backprop` in `multilayer_perceptron.cc`
* `event_times_buffer` in `diagram.cc`
* `log` in `vector_log_sync.cc`
* `scratch` in `rgbd_sensor_async.cc`