RobotLocomotion/drake #15950: sdformat: Should allow URDF files to be included via include tags

EricCousineau-TRI, 2021-10-20:
Steps:

- Ensure we have negative test showing that URDF inclusion via include tag fails *fast* (rather than allowing SDFormat to convert)
- Ensure we have necessary changes to libsdformat interface API to support including URDF files (without loss of information)

jwnimmer-tri, 2022-04-19:
I see that the ignitionrobotics pull request has been merged.  What is the next step to resolve this issue?

Model composition is an important feature in Drake.  I'd like to advocate that we prioritize OpenRobotics to _finish_ the model-includes feature that has been an ongoing project for several years,  before we start working on _new_ projects.

EricCousineau-TRI, 2022-04-19:
There is still pending work for this as tracked on SDFormat project board:
https://github.com/orgs/ignitionrobotics/projects/6

Outstanding work:
- Ensure that our usage of Interface API accommodates all of Drake's parseable model types. See https://github.com/RobotLocomotion/drake/pull/16727
- Ensure we have conversion scripts for model directives to SDFormat composition. Will require redux of https://github.com/aaronchongth/drake/pull/1

rpoyner-tri, 2026-11-12:
Closing this as done. Both steps are covered by Drake's current SDFormat parser.

**Interface API support for URDF without loss (step 2).** #16727 added merge-include support for custom-parsed models. Drake now handles any included `.urdf` or `.xml` file itself, in `ParseNestedInterfaceModel` in `multibody/parsing/detail_sdf_parser.cc`.
- It sends the file to Drake's own parser, so libsdformat's URDF-to-SDFormat conversion is bypassed.
- For `merge="true"` includes it calls `MergeModel` on the parent model instance.
- It reports the result back to libsdformat through the Interface API, with `SetParserSupportsMergeInclude(true)`.

**Fail-fast negative test (step 1).** No longer needed. Drake claims every included `.urdf` before libsdformat can convert it, so a merge-included URDF can't be silently converted. Merge and non-merge URDF includes are both covered in `multibody/parsing/test/detail_sdf_parser_test.cc` (`InterfaceApi`, `ErrorsFromIncludedUrdf`, and the merge-parameterized joint and frame tests). That test target passes at current HEAD.

Two items from the earlier comments are not covered by this issue:
- Conversion from model directives to SDFormat composition is tracked in #15948.
- The canonical-link behavior for models from included URDF or MJCF files is tracked in #25129.

closed 2026-11-12 by rpoyner-tri