RobotLocomotion/drake #16025: Support for vectorized constraints in MathematicalProgram

buoyancy99, 2021-11-01:
In the python API of any optimization software (e.g.cxvpy), we can add simple constraints in a vectorized way.

For example, in cvxpy we can do something like
```
x = cvxpy.Variable(3)
constraints = [0 <= x, x <= 1]
```

I observed that drake doesn't support vectorized constraints in such simple grammar. To achieve the same thing we can only manually convert a problem to form Ax<=b and then use `AddLinearConstraint(A, lower, upper, x)` or `AddBoundingBoxConstraint(lower, upper, x)`.

Consider the 2-dim case, it would be nice for drake to support grammar like
```
prog = MathematicalProgram()
X = prog.NewContinuousVariables(2, 2, "X")
prog.AddLinearConstraint(np.sum(X, 1) >= np.array([0.5, 0.5]))
prog.AddLinearConstraint(X >= 0)
```
which is supported in cvxpy. 

Any plan to implement this in the future?

hongkai-dai, 2021-11-01:
You can do
```
prog.AddLinearConstraint(np.sum(X, 1), np.array([0.5, 0.5]), np.array([np.inf, np.inf]))
```
It is not as readable as `np.sum(X, 1) >= np.array([0.5, 0.5])`, but it is better than figuring out `A` matrix and then do AddLinearConstraint(A, lb, ub, x).

jwnimmer-tri, 2021-11-02:
We have a little trouble integrating Formula with numpy for now; see #8315.

For the moment, my only suggestion is to add the constraints using the expressions that you want, but spelling out the list comprehensions manually (instead of relying on array broadcasting):

```python
prog = mp.MathematicalProgram()
X = prog.NewContinuousVariables(2, 2, "X")
prog.AddConstraint([term >= 0.5 for term in np.sum(X, 1)])
prog.AddConstraint([term >= 0.2 for term in np.sum(X, 0)])
prog.AddConstraint([term >= 0.0 for term in X.flatten()])
for c in prog.GetAllConstraints(): print(c)
```

Output:
```
LinearConstraint
0.5 <= (X(0,0) + X(0,1)) <= inf
0.5 <= (X(1,0) + X(1,1)) <= inf

LinearConstraint
0.2 <= (X(0,0) + X(1,0)) <= inf
0.2 <= (X(0,1) + X(1,1)) <= inf

BoundingBoxConstraint
0 <= X(0,0) <= inf
0 <= X(0,1) <= inf
0 <= X(1,0) <= inf
0 <= X(1,1) <= inf
```

Once we fix #8315, I suspect that this would work:
```python
prog = mp.MathematicalProgram()
X = prog.NewContinuousVariables(2, 2, "X")
prog.AddConstraint(np.sum(X, 1) >= 0.5)
prog.AddConstraint(np.sum(X, 0) >= 0.2)
prog.AddConstraint(X >= 0.0)
```

RussTedrake, 2021-11-03:
I'm late to the party (and excited if we can make progress on this!).  Based on https://github.com/RobotLocomotion/drake/issues/8315#issuecomment-480773969, I believe we can also write
```
prog = mp.MathematicalProgram()
X = prog.NewContinuousVariables(2, 2, "X")
prog.AddConstraint(ge(np.sum(X, 1),0.5))
...
```
already...

jwnimmer-tri, 2021-11-03:
Right!  I knew I was forgetting something.

```python
import numpy as np
from pydrake.all import *
prog = MathematicalProgram()
X = prog.NewContinuousVariables(2, 2, "X")
print(prog.AddConstraint(ge(np.sum(X, 1),0.5)))
```

```console
LinearConstraint
0.5 <= (X(0,0) + X(0,1)) <= inf
0.5 <= (X(1,0) + X(1,1)) <= inf
```