RobotLocomotion/drake #16040: Compliant-compliant hydroelastic contact

DamrongGuoy, 2021-11-04:
This is the implementation plan to change prototype #15545 into quality code in master.  It will be updated as we go. Each item should become a small PR.

#### ComputeContactSurfaceFromCompliantVolumes()
  - [ ] Equilibrium plane between two linear functions from two tetrahedra.
  - [ ] Intersection of a plane and a tetrahedron
  - [ ] Intersection of a polygon and a few halfspaces 
    - Refactor SurfaceVolumeIntersector::ClipPolygonByHalfSpace() for sharing between rigid-compliant and compliant-compliant mesh intersection.
  - [ ] Heuristic to filter out special cases
    - Only allow contact polygons whose normal vectors are along pressure gradients. 
  - [ ] Contact polygon between two tetrahedra with pressure fields
  - [ ] Contact polygons between two tetrahedral meshes with pressure fields
  - [ ] Contact surface between two compliant hydroelastic tetrahedral meshes
 
#### ComputeContactSurfaceFromCompliantHalfspaceCompliantMesh()
  - [ ] Equilibrium plane between a tetrahedron's pressure field and the halfspace's pressure field.
  - [ ] Intersection of a plane and a tetrahedron (shared with the previous section)
  - [ ] Intersection of a polygon and the halfspace (shared with the previous section)
  - [ ] Heuristic to filter out special cases (shared with the previous section)
  - [ ] Contact polygon between a tetrahedron with pressure field and a halfspace with pressure field
  - [ ] Triangle(or polygon) mesh from a tetrahedral mesh with pressure field and a halfspace with pressure field
  - [ ] Contact surface between a compliant hydroelastic halfspace and a compliant hydroelastic tetrahedral mesh

#### Use the new capability in SceneGraph
  - [ ] Update hydroelastic_callback
  - [ ] Update doxygen of QueryObject::ComputeContactSurfaces() 

#### Use the new capability in MultibodyPlant
  - [ ] Update MultibodyPlant::CalcDiscreteContactPairs()
    - Combine normal pressure gradients from two bodies
  - [ ] Confirm with @amcastro-tri that it works well with HydroelasticTractionCalculator and other parts.

DamrongGuoy, 2021-12-13:
@hongkai-dai 

I have to postpone compliant-compliant hydroelastic contact to CY22Q1 since #15796 has a higher priority.  This issue tracks the progress of compliant-compliant hydroelastics.

closed 2022-02-02 by DamrongGuoy

sherm1, 2022-02-02:
Congratulations @DamrongGuoy & team! 👏👏👏