RobotLocomotion/drake #16043: Specify Geometry Role in SDF visual tag.

TheCynosure, 2021-11-04:
I have an SDF that has two possibly models that represent it at different levels of detail. I would like to be able to specify one as an Illustration model and another as a Perception model. This way I can have a lower detail model for the visualizer and higher detail model available for my sensors.

SeanCurtis-TRI, 2021-12-04:
Here's the shape of the basic solution:

We currently support the `<drake:proximity_properties>` custom tag for both sdf and urdf. It has always been the intent of providing `<drake:illustration_properties>` and `<drake:perception_properties>` as well. Adding these missing tags is the crux of the solution.

Special notes:
  - The current default behavior is that a `<visual>` geometry is used for both illustration and perceptions roles. This should remain the default behavior if neither of the drake role tags are provided.
  - For the diffuse rgba value, for the files that allow definition of *materials*, the rgba value should still be configurable by referencing a defined material.

jwnimmer-tri, 2022-03-09:
See also #13689 and #11746.

jwnimmer-tri, 2023-05-23:
> See also https://github.com/RobotLocomotion/drake/issues/13689 and https://github.com/RobotLocomotion/drake/issues/11746.

Those tickets have more information, and as far as I can tell it's the identical request.  Closing this a duplicate of those.

closed 2023-05-23 by jwnimmer-tri