RobotLocomotion/drake #16069: pip install: Parser.AddModelFromFile - Unable to find `drake/package.xml`?

EricCousineau-TRI, 2021-11-10:
Was trying to write up example (inline def. of SDFormat file, parsing it, etc.), but found the following error (min. repro for Ubuntu 18.04, CPython 3.6.9):

```sh
$ python3 -m venv .
$ source ./bin/activate
$ pip install -U pip wheel
$ pip install drake==0.35.0

$ python3 <<EOF
from pydrake.all import MultibodyPlant, Parser
Parser(MultibodyPlant(0.0)).AddModelFromFile("/this/file/does/not/exist")
EOF

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Could not find Drake resource_path 'drake/package.xml' because Drake CMake install marker specified a resource root of '/tmp//lib/python3.6/site-packages/pydrake/lib/../share' but that root did not contain the expected file '/tmp//lib/python3.6/site-packages/pydrake/lib/../share/drake/package.xml'.
```

jwnimmer-tri, 2021-11-10:
Oh.  The `package.xml` file isn't even in the wheel.  That would do it!  The wheel is also missing the `geometry/**` meshcat files.  I think the build-wheel script should default to copying all shared resources, with a denylist for the stuff we don't want (resource_tool, repo.bzl, install_prereqs), rather than trying to keep an allowlist up to date.

jwnimmer-tri, 2021-11-10:
(High priority, since I think nobody can use sdformat nor urdf files at all without this.)

jwnimmer-tri, 2021-11-10:
For now, the work-around for users would be to copy that [package.xml](https://github.com/RobotLocomotion/drake/blob/v0.35.0/package.xml) file to `python3.6/site-packages/pydrake/share/drake/package.xml` by hand.

@mwoehlke-kitware if you have time today to PR a fix here, that would be helpful.  Otherwise, I'll get to it at the end of the day.

closed 2021-11-10 by jwnimmer-tri

jwnimmer-tri, 2021-11-16:
This fix was released in `drake==0.36.0`.

PR #16071 (2021-11-10): Add missing resources to wheel
Fixes #16069.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16071)