RobotLocomotion/drake #16098: SceneGraph's documentation doesn't tell me why I want to name my renderer

RussTedrake, 2021-11-15:
I've always found it a bit odd that renderers are keyed by string names, and don't understand why.  I was just reading through the docs on `SceneGraph` and `RenderEngine`, but didn't find the answer.

In particular, a user (me, this time) would really like to know:
- Would I ever register multiple renderers of the same type?
- Is it ok for multiple `RgbdSensor`s to use the same renderer?  (I've been doing it)
- Would there ever be a performance benefit to having different cameras using different renderers?
- Would that answer change if the cameras were rendering in different threads?

SeanCurtis-TRI, 2021-12-04:
Note for when resolving this:

The motiviation behind naming a renderer is so that custom renderers could be introduced. For example, one might generate a special render for doing amazing simulation of a particular depth sensor (but not do label or rgb). So, to produce color and labels, two renderers would have to be added. They get named so as to be distinct. 

Renderer names can also be used if you want to target specific geometries to specific renderers (using the `<drake:accepting_renderer>` tag in URDF or SDF. In addition to having different geometries that have been created for specific renderers, this tag can be used to omit `<visual>` geometry from *any* renderer (making it just serve an illustration role) by using a name that doesn't map to any renderer at all.

jwnimmer-tri, 2023-04-06:
As a helpful cross-reference for others who stumble here:

We recommend using [pydrake.systems.sensors.CameraConfig](https://drake.mit.edu/pydrake/pydrake.systems.sensors.html#pydrake.systems.sensors.CameraConfig) and its helper function `ApplyCameraConfig` to add your cameras.

It has a property for `renderer_class` to specify which built-in engine to use and will automatically create the engine the first time its needed, and reuse it thereafter for any additional cameras that want the same class of engine.