RobotLocomotion/drake #16099: [meshcat] Port StartMeshcat to Drake

RussTedrake, 2021-11-15:
@rpoyner-tri ... you probably recall the PR where you didn't want Meshcat in C++ to be able to advertise unconfirmed URLs.  My work-around at that time was the `StartMeshcat` python method in manipulation:

https://github.com/RussTedrake/manipulation/blob/b6d32c075dd78d268580b098378ccdb008640ec0/manipulation/meshcat_cpp_utils.py#L27

As we start to encourage the use of Drake+Meshcat on deepnote/colab, this is missing functionality; not a good reason to add a dependency on the entire manipulation repo.

Do you have any objections to moving this (python-only) method to Drake?  I believe it instantiates your previous recommendation, but want to confirm before I open the PR.

rpoyner-tri, 2021-11-15:
Looks ok to me -- right abstraction level, and doesn't require any new dependencies that I can detect. I'm not sure where it should live, but no doubt @EricCousineau-TRI and @jwnimmer-tri will have ideas.

EricCousineau-TRI, 2021-11-15:
No objections here!

W.r.t. layout, `pydrake.geometry.jupyter` sounds good to parallel other cases (`pydrake.common.jupyter`). Two observations, though:
- With current file layout, I believe placing module in `pydrake.geometry.jupyter` will still require the dependency be present when importing `pydrake.geometry` (since top-level is C++).
- IPython should be present by default, given that we depend on `meshcat-python`? https://github.com/rdeits/meshcat-python/blob/v0.3.2/setup.py#L21

jwnimmer-tri, 2021-11-15:
> IPython should be present by default, given that we depend on meshcat-python?

We will soon be deprecating our dependency on `meshcat-python`.

RussTedrake, 2021-12-21:
N.B. -- in #16273, I've removed any reference to IPython.  In fact, it's not even specific to Jupyter; it could be run from a Deepnote terminal in a python script.

PR #16273 (2021-12-21): [meshcat] Provides StartMeshcat() for Deepnote/Colab
Resolves #16099

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16273)
commit 5293e8e1 (2021-12-21): [meshcat] Provides StartMeshcat() for Deepnote/Colab (#16273)

Resolves #16099
closed 2021-12-21 by RussTedrake