RobotLocomotion/drake #16105: Document new build-wheels script

mwoehlke-kitware, 2021-11-16:
Per #16094, we need to add more detailed documentation on how to use the new Python `build-wheels` script, probably to an accompanying (i.e. in `tools/wheel`) README.

We should also consider migrating some of the "how to release wheels" instructions to this new document.

jwnimmer-tri, 2021-12-13:
Pasting some information from the PR review thread, to help focus and explain this issue:

---
  
> For all of the arguments other than "output-dir", we need to document for other developers the purpose(s) that they should want to use these arguments for.
>
>It's possible that more text in the helpstring would suffice. (For example, with "keep containers" maybe the helpstring goes on to say "This can be useful to help inspect intermediate results for debugging. If you've set this option, then you can type 'such and such' in order to launch an interactive bash shell in the container".)
>
>If it's too much information to fit into the strings here, then a README.md would be another good alternative.
>
>The goals we need to meet are twofold:
>
> (1) To educate other developers on some tips and best practices for using this tool.
>
> (2) To provide concrete advice on what it looks like to successfully use the option, so that when someone comes along to propose a change to this file, they can test the change locally w.r.t. all of the options that are relevant to the change, i.e., that developers are able to confidently propose modifications to this file.

---

> -k: Hopefully we both know what this does, but agreed it could use documentation.
>
> -s: Generates a bunch of (retained) tags like pip-drake:clean-ubuntu-py37 for the various stages of builds. Useful for poking around after the fact. (Indeed, my "main" goal with this rewrite was to integrate this functionality, which I currently maintain as a separate, not committed script.)

---

> Most of these are pretty clear on the "what" (which is sufficient for a --help string). What a bunch of them lack is the "why" or an overall "how" in the bigger picture.
>
> For the example of -n, the "what" is already clear. But why should anyone care about this in the first place? Under what circumstances would a future developer or user of this script benefit from setting this option? I can't yet see myself ever specifying it.
> 
> Similarly for -k and -s. You seem to have some local workflow where you use those options in earnest, to help accelerate your development or testing cycles. Given the lack of documentation that explains how to apply those options effectively in some example situations, nobody else will be able to similarly accelerate their local development. To that end, I'm less worried about other people's effectiveness here, but more worried that if nobody else knows how those options are supposed to be effectively applied, then nobody else will ever use or test them, and so they will bitrot.
>
> Admittedly trying to write down the "how / why" in comments here is might be much to ask, thus the option of leaving a README / TODO for the future. As you note, we might also incorporate the whl playbook text into a README here as well, and mothball the online document. I'm OK with all of that in a future PR, with either a TODO or Drake issue to remind us about it.

---

> I use -sn (well, used my old build-images.sh script) all the time during testing, as a "run and forget" when I'm trying to debug build problems. It's useful to not clutter my filesystem with build artifacts. Once a build succeeds, I'll worry about actually extracting and testing the wheels. (And I also use it just to "freshen" my -s tagged images, so that they're closer to up-to-date next time I have to go poking around the install tree because the wheel is missing some artifact, e.g. the recent package.xml.)
>
> -k, similarly, is for debugging a failed build... although TBH, I've never used it, but it could actually save a lot of time if used habitually. The down side is it leaves artifacts any time a build fails, but if you have those artifacts, you can jump into debugging much, much faster. Anyway, its implementation cost is negligible (it implements "Remove --force-rm if you need to inspect artifacts of a failed build" from the old script). Given that all it does is suppress building with --force-rm, I'm not too worried about bitrot on that.
>
> I think I'd be in favor of migrating all of this documentation to tools/wheel[/dev]/README[.*]. Want me to open an issue for that? (Is there anything currently in the README that shouldn't be public knowledge? The only thing that comes to mind is if we're too specific about where the drake-robot credentials live... since I expect that "user" to still exist and be active, and since we're discouraging its use outside CI, I think mentioning that it exists should be fine...)

jwnimmer-tri, 2022-01-14:
Okay, the call to action here is as follows:

- Open a PR with a `drake/tools/wheel/README.md`:
  - ... with a brief summary of the purpose of this folder (to build pypi wheels for monthly releases)
  - ... with a link to https://drake.mit.edu/release_playbook.html as reminder of the release process.
  - ... with a best-effort, first pass at maintainer documentation for developers that will be modifying the wheel builds.  We have various suggestions for this already upthread in this issue; the work here is to boil that down into a basic attempt at a README.  If people encounter maintenance challenges in the future, we can always work to add more README content down the road.

PR #16417 (2022-01-24): Add README for build-wheels
Add a brief document explaining the build-wheels tool. This includes both some basic usage information, particularly more detailed descriptions and rationale for some of the arguments, as well as some brief information on the script's internal operation.

Fixes #16105.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16417)
closed 2022-01-24 by jwnimmer-tri