RobotLocomotion/drake #16109: Objects missing in MeshCat visualizer

AndyMarchese, 2021-11-16:
I encountered an issue where C++ MeshCat only drew a subset of objects within a scene; notably, these scenes included a large number of objects (25+) having mesh geometries and textures. The backpressure limit warning [here](https://github.com/RobotLocomotion/drake/blob/75e949b9dba02d650e7f5560adb2b468c8a82b1e/geometry/meshcat.cc#L84-L92) was not tripped. As Russ calls out in the comments [here](https://github.com/RobotLocomotion/drake/blob/75e949b9dba02d650e7f5560adb2b468c8a82b1e/geometry/meshcat.cc#L1189-L1192), the problem appears to be that the message queue fills up and some `set_object` messages are missed. To quickly confirm this, I used a sleep to slow down the calling thread:
```
std::this_thread::sleep_for(std::chrono::milliseconds(1000));
publisher_->SetObject(path, shape, rgba);
```
This produced the correct behavior - all geometries were drawn as expected. But as suggested in the comment, the right fix is to probably throttle the calling thread based on buffer size.

RussTedrake, 2021-11-20:
One subtlety here, which I want to register to make sure people understand/approve.  I do think the fundamental solution is to slow down the main simulation thread so that the communications thread has time to send it's messages.  But this means that if there are multiple browsers connected to the meshcat scene, the simulation will be slowed down to the _slowest_ connection.  Are we sure that's what we want?

SeanCurtis-TRI, 2021-11-29:
I'm a little confused. I wouldn't expect that `SetObject` would be called extensively in the simulation loop. That's what registers objects into meshcat, right? The simulation loop, I would imagine, largely consists of calls to `SetTransform`. While the message arising from a `SetObject` invocation can be arbitrarily terrifyingly large, I'd imagine that `SetTransforms` would be quite compact. So, before we "slow down the main simulation thread", are we actually solving the problem? It seems what needs to be queued/throttled is the initialization.

Is there a way that the `SetObject` methods can detect if the queue is empty/full and only submit to an "empty" (or "near" empty) queue? Alternatively, if we already embrace the idea of "slowing down" the simulation to some heuristically-derived limit, can't we just embed that in the call to `SetObject` so that it blocks slightly before returning? Sure, it slows down initialization, but lets the actual simulation run at normal speeds.

Again, all of this is based on my mental model about how much I would expect to see `SetObject` called during any meaningful simulation step.

SeanCurtis-TRI, 2021-11-29:
Nevermind. I see that #16144 has gone a completely different direction. Please ignore my comment.

PR #16144 (2021-11-30): [meshcat] Avoid dropping messages due to backpressure, and provide Flush()
Resolves #16109

After reading through the uwebsockets code, I understand now that the
maxBackpressure websocket option was *only* being used to decide when
to drop packets.  We never want to drop packets, so setting that to
zero is the correct behavior.

Backpressure is still a legitmate concern.  If we keep shoving data
into the send buffer faster than the websocket can send, then
backpressure will build up (consuming memory and causing the
visualizer to fall behind the main thread).  I've provided a `Flush()`
command [... cut]
commit ba9f8b78 (2021-11-30): [meshcat] Avoid dropping messages due to backpressure, and provide Flush() (#16144)

* [meshcat] Avoid dropping messages due to backpressure, and provide Flush()

Resolves #16109

After reading through the uwebsockets code, I understand now that the
maxBackpressure websocket option was *only* being used to decide when
to drop packets.  We never want to drop packets, so setting that to
zero is the correct behavior.

Backpressure is still a legitimate concern.  If we keep shoving data
into the send buffer faster than the websocket can send, then
backpressure will build up (consuming memory and c [... cut]
closed 2021-11-30 by RussTedrake