RobotLocomotion/drake #16116: Inscrutable error message when two different MbPs share contact results

jwnimmer-tri, 2021-11-17:
Inspired by a [StackOverflow posting](https://stackoverflow.com/questions/69947778/contact-between-two-models-causing-indexerror-map-baseat): if a user connects one SceneGraph to two different MultibodyPlants, I believe it will mostly work.  I think that cameras would see both plants' geometries, all would be visualized, etc.  However, I think if bodies from different plants ever make contact, then we cannot solve the dynamics.  Currently, we throw a `IndexError: _Map_base::at` with no other message in this case.  Ideally, we would report a more sensible diagnostic.

jwnimmer-tri, 2021-11-17:
\CC @sherm1 @SeanCurtis-TRI @rpoyner-tri FYI

The specific failure was in `CalcNormalAndTangentContactJacobians`, as I recall.  In any case, I think the take-away here is that data coming from input ports should be "untrusted" in the same way that function arguments on public API are untrusted -- we should sanity check them before running off and using them.

Thus, I think that lookups in `geometry_id_to_body_index_` based on contact results inputs should be guarded by nice exception checkers.  Possibly some other lookup map accesses will also need guards.  Some helper function(s) are probably in order.

SeanCurtis-TRI, 2021-11-17:
In my mind this problem manifests two separate problems: a superficial one and a fundamental one.

Superficial: `MultibodyPlant` has too much information about contact geometry. It has this information for anachronistic reasons. About a year ago, and initial attempt was started to disentangle it. I don't recall why it wasn't brought across the finish line. @joemasterjohn probably has the best insight into that.

Fundamental: separation of `MultibodyPlant` and `SceneGraph` inherently opens the door to geometries that `MultibodyPlant` knows nothing about. Even if it relies on `SceneGraph` to inform it of the geometric properties of an apparent contact, it still has to resolve the dynamics. `MultibodyPlant` can only impart forces on the bodies it controls. What should it do? Simply throwing (with an intelligible message) would be sufficient to fail gracefully. This has been lurking in the back of my mind since 2017.

amcastro-tri, 2021-11-17:
I agree with @SeanCurtis-TRI's observations above, I think throwing an exception is the right solution at this point.

jwnimmer-tri, 2022-01-15:
> Superficial: MultibodyPlant has too much information about contact geometry. It has this information for anachronistic reasons. About a year ago, and initial attempt was started to disentangle it. I don't recall why it wasn't brought across the finish line.

Actually, a lot of the bookkeeping is dead code.  I remove the easy parts in my PR fix for this.

PR #16390 (2022-01-19): [multibody] Guard hazardous map access by GeometryId
The GeometryId values for collisions come in on an input port. We need to sanity-check them prior to using them.

For two maps that were already dead code, we just kill the deadwood instead of adding guards for it.  The related `default_coulomb_friction_` was also dead code, so is also purged.

Closes #16116.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16390)
closed 2022-01-19 by jwnimmer-tri