RobotLocomotion/drake #16125: can we integrate libcurl for the render client?

svenevs, 2021-11-18:
CC: @jwnimmer-tri @mwoehlke-kitware 

- License validation.
- Dependencies:
    - zlib
    - zstd (?)
    - TLS of some kind, default is libopenssl. Can also be disabled.

Where pip wheel builds are concerned, the CMake build had some strange link order problem with `zstd` but the autotools build was fine.  Did not investigate closely, but should be fixable.

jwnimmer-tri, 2021-11-18:
Yes, the curl library license is fine.  It is some blend of MIT-like and BSD and ISC.

The ROS Noetic install on Ubuntu Focal forces `libcurl4-openssl-dev` to be used -- it doesn't respect the user by depending only on the virtual package `libcurl4-dev` like it should be.  (I'll file an upstream ROS bug about this, eventually.)

Therefore, at least for now, Drake's `install_prereqs` for the `source_distribution` should use the `libcurl4-openssl-dev` spelling, and the `binary_distribution` should use `libcurl4`.

I think the wheel build could use whatever flavor is easiest.  That is probably mean disabling TLS in the wheel build.

jwnimmer-tri, 2021-11-18:
Actually, if possible, doing `sudo apt-get satisfy 'libcurl4-gnutls-dev | libcurl4-dev'` would be better than hard-coding a specific one.

Possibly that means our install script should change from using `apt-get install` to using `apt-get satisfy` instead, so that we can put the disjunction(s) directly into our `packages-{dist}.txt` files.  Or maybe our install script should continue to use `install` for any boring lines, but use `satisfy` for any lines with non-name chacters (`<>=|`).

svenevs, 2021-11-29:
I'm running into a shell quoting problem that I don't know how to approach to get `apt-get satisfy` working.  Ideally, `packages-${codename}.txt` should have the correctly-quoted string inside of it, e.g., 

```diff
--- a/setup/ubuntu/source_distribution/packages-focal.txt
+++ b/setup/ubuntu/source_distribution/packages-focal.txt
@@ -10,6 +10,7 @@ git
 libblas-dev
 libbz2-dev
 libclang-9-dev
+'libcurl4-gnutls-dev | libcurl4-dev'
 libdouble-conversion-dev
 libexpat1-dev
 libgflags-dev
```

The way I'm trying to parse things:

```bash
packages=$(cat "${BASH_SOURCE%/*}/packages-${codename}.txt")
satisfy_charset="[<>=|]"
install_packages="$(grep -v "$satisfy_charset" <<< "${packages}")"
satisfy_packages="$(grep "$satisfy_charset" <<< "${packages}")"
$apt_get_install --no-install-recommends ${install_packages}
set -x
echo "satisfy packages:"
echo ${satisfy_packages}
$apt_get_satisfy ${satisfy_packages}
```

producing

```
++ echo 'satisfy packages:'
satisfy packages:
++ echo ''\''libcurl4-gnutls-dev' '|' 'libcurl4-dev'\'''
'libcurl4-gnutls-dev | libcurl4-dev'
++ apt-get satisfy ''\''libcurl4-gnutls-dev' '|' 'libcurl4-dev'\'''
E: Problem parsing dependency: |
```

It's desirable to have the correctly quoted satisfy argument in the file, since the [example here](https://ubunlog.com/en/lista-la-nueva-version-de-apt-2-0-y-estas-son-sus-novedades/) `apt-get satisfy "foo" "Conflicts: bar" "baz (>> 1.0) | bar (= 2.0), moo"` is quite complex, trying to quote after (e.g. with `sed`) would not be robust.

Any thoughts?  I've exhausted all of my shell skills trying to work around this while still also supporting potentially more than one of these ending up in `packages-${codename}.txt`.

jwnimmer-tri, 2021-11-29:
First, the important risk to investigate here is whether Render RPC or libcurl work at all ever, not how we install the prereqs.  For prototyping, just hard-code `libcurl4-gnutls-dev` and move on to the important stuff.   We can fix the prereqs spelling during pull request review / iteration later on.

Second (moving on to the future work of the actual prereqs code that we'd want to merge to master), I'd start by replacing `$apt_get_install` with `apt-get install ${maybe_yes}` to remove a layer of indirection.  We only need the `-y` to be variable-expanded, not the whole command.

Third, I'd suggest placing the satisfy packages into a separate text file; this will remove the need to grep -- which will be easier for downstream projects like Anzu that consume the text files directly, so they don't have to re-implement the grep part.

Finally, for shell stuff I'd just use `xargs`.  Something like `cat packages-satisfy.txt | xargs apt-get satisfy`.

PR #16233 (2021-12-21): add libcurl dependency
For communicating with the render server.

Fixes #16125.

What do we want to do about the `apt-get satisfy` scenario?

Relates: #16234

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16233)
closed 2021-12-21 by sherm1