RobotLocomotion/drake #16146: Spatial velocity clarification

bostoncleek, 2021-11-28:
Hi,

I have a question regarding the function [`EvalBodySpatialVelocityInWorld()`](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html#a8377774b012b4975b46941b3197f29c3). 

It seems the linear component produced by this function is not the spacial velocity but is instead the 
velocity of the body frame relative to the world  (below this is `pdot`). Can you verify this functions is returning the spatial velocity and not the classical mechanics definition of velocity `pdot`.

[image]

hongkai-dai, 2021-11-28:
That is right. The linear component of a spatial velocity is the velocity of the body relative to the world, namely pdot.

The v_ab^s term  in your equation is probably the linear component of a twist, which is the velocity of a point fixed to the body but instantaneously coincides with the world origin. Featherstone described the kinematics and dynamics using twist. On the other hand Drake doesn't use twist, but spatial velocity instead.

bostoncleek, 2021-11-28:
Ok thank you

closed 2021-11-28 by bostoncleek