RobotLocomotion/drake #16148: Adding a doxygen comment with the old-style @system tag causes mkdoc to hang / give no hints.

RussTedrake, 2021-11-29:
To reproduce, add 
```
 @system{MeshPainterSystem,
   @input_port{geometry_query},
   @output_port{texture}
 }
```
into the class overview doxygen string of any installed header file.

On my machine, the `bazel build` step creating `documentation_pybind.h` eventually terminated after 200+ seconds with effectively no error message.  Adding `--sandbox_debug` didn't help.

The newer syntax, e.g.
```
@system
name : MeshPainterSystem
input_ports:
- geometry_query
output_ports:
- texture
@endsystem
```
works as expected.  The only issue here is that mkdoc falls down as hard as it does on the old syntax.  

I tripped over this during a port from anzu; by my quick count there are 7 instances of the old syntax still hanging around in anzu master.  If the resolution here is "fix the 7, and treat the future failures as nearly 0% probability", then I'm ok with that.  Sorry if it's _my_ mkdoc-related code that has turned out to be brittle!

rpoyner-tri, 2021-11-29:
To begin, we have this python stack trace from a run of ` bazel build //bindings/pydrake:generate_pybind_documentation_header`:

```
Traceback (most recent call last):
  File "/home/rico/.cache/bazel/_bazel_rico/7f8997f28c9253517a55d673c67a6c74/sandbox/linux-sandbox/5/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/mkdoc.runfiles/drake/tools/workspace/pybind11/mkdoc.py", line 858, in <module>
    main()
  File "/home/rico/.cache/bazel/_bazel_rico/7f8997f28c9253517a55d673c67a6c74/sandbox/linux-sandbox/5/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/mkdoc.runfiles/drake/tools/workspace/pybind11/mkdoc.py", line 828, in main
    extract(include_file_map, translation_unit.cursor, symbol_tree)
  File "/home/rico/.cache/bazel/_bazel_rico/7f8997f28c9253517a55d673c67a6c74/sandbox/linux-sandbox/5/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/mkdoc.runfiles/drake/tools/workspace/pybind11/mkdoc.py", line 328, in extract
    extract(include_file_map, i, symbol_tree, deprecations)
  File "/home/rico/.cache/bazel/_bazel_rico/7f8997f28c9253517a55d673c67a6c74/sandbox/linux-sandbox/5/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/mkdoc.runfiles/drake/tools/workspace/pybind11/mkdoc.py", line 352, in extract
    extract(include_file_map, i, symbol_tree, deprecations)
  File "/home/rico/.cache/bazel/_bazel_rico/7f8997f28c9253517a55d673c67a6c74/sandbox/linux-sandbox/5/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/mkdoc.runfiles/drake/tools/workspace/pybind11/mkdoc.py", line 352, in extract
    extract(include_file_map, i, symbol_tree, deprecations)
  File "/home/rico/.cache/bazel/_bazel_rico/7f8997f28c9253517a55d673c67a6c74/sandbox/linux-sandbox/5/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/mkdoc.runfiles/drake/tools/workspace/pybind11/mkdoc.py", line 358, in extract
    comment = process_comment(comment)
  File "/home/rico/.cache/bazel/_bazel_rico/7f8997f28c9253517a55d673c67a6c74/sandbox/linux-sandbox/5/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/mkdoc.runfiles/drake/tools/workspace/pybind11/mkdoc_comment.py", line 68, in process_comment
    s = process_doxygen_commands(s)
  File "/home/rico/.cache/bazel/_bazel_rico/7f8997f28c9253517a55d673c67a6c74/sandbox/linux-sandbox/5/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/mkdoc.runfiles/drake/tools/workspace/pybind11/mkdoc_comment.py", line 259, in process_doxygen_commands
    s = process_doxygen_to_sphinx(s)
  File "/home/rico/.cache/bazel/_bazel_rico/7f8997f28c9253517a55d673c67a6c74/sandbox/linux-sandbox/5/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/mkdoc.runfiles/drake/doc/doxygen_cxx/system_doxygen.py", line 120, in process_doxygen_to_sphinx
    return _process_doxygen(s, system_yaml_to_pydrake_system_rst_directive)
  File "/home/rico/.cache/bazel/_bazel_rico/7f8997f28c9253517a55d673c67a6c74/sandbox/linux-sandbox/5/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/mkdoc.runfiles/drake/doc/doxygen_cxx/system_doxygen.py", line 106, in _process_doxygen
    s = s[:start] + transformed + s[end + len(end_tag):]
MemoryError
```

And, yes, sandbox_debug is likely to be of any help. Getting a command line extracted outside of bazel is a likely next step.

jwnimmer-tri, 2021-11-29:
Seems like the line `end = s.find(end_tag, start + len(start_tag))` lacks any error checking for when `@endsystem` is missing (the `find()` returns -1, which confuses the index arithmetic that follows).

rpoyner-tri, 2021-11-29:
Meh, I found it quite difficult to reverse-engineer a tolerable invocation of `mkdoc`. In the end the traceback and code reading were all that was really necessary. See PR coming shortly.

PR #16152 (2021-11-30): doc: Improve parsing and error reporting for @system diagram syntax
Closes: #16148

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16152)
commit cac0c83b (2021-11-30): doc: Improve parsing and error reporting for @system diagram syntax (#16152)

* doc: Improve parsing and error reporting for @system diagram syntax

Closes: #16148
closed 2021-11-30 by rpoyner-tri