RobotLocomotion/drake #16178: supported and custom tags for SDF proximity properties are undiscoverable

RussTedrake, 2021-12-04:
I just got a perfectly reasonable question: "How do I set the friction coefficient in an SDF".  It was surprisingly hard to find the answer!

My answer was: For controlling the friction properties, you can add additional tags to the sdf:
https://github.com/RobotLocomotion/drake/blob/master/multibody/parsing/detail_scene_graph.h#L159
as seen in https://github.com/RobotLocomotion/drake/blob/62750b830658b2013113ce26e9f9b56327e945c4/bindings/pydrake/multibody/test/hydroelastic.sdf#L26

But all that beautiful documentation in `detail_scene_graph.h` is under an `internal` scope, and does not appear in any user documentation.  Unless I'm missing something?

SeanCurtis-TRI, 2021-12-04:
Closing as redundant of #13314.

RussTedrake, 2021-12-04:
Fair.  #13314 suggests that it is only hydroelastic parameters that are hidden.  But this is a problem for all proximity parameters (including point contact); and really all parsing documentation.

SeanCurtis-TRI, 2021-12-04:
Surely #13314 refers to many things that have nothing to do with hydroelastic:

  - textures not appearing in camera
  - collision filters
  - reflected inertia parameters
  
Other than the fact that two of the examples refer to hydroelastic, there is literally no language that refers to it. Still, if you're dying for this issue to remain open. I won't stand in the way.

rpoyner-tri, 2022-03-18:
I will claim this need was largely satisfied by #16444. Of course there is more work to do, but not enough to justify keeping this issue.

closed 2022-03-18 by rpoyner-tri