RobotLocomotion/drake #16180: update effort limits in iiwa description files?

RussTedrake, 2021-12-05:
Someone pointed out to me that the effort limits in our [sdf](https://github.com/RobotLocomotion/drake/blob/master/manipulation/models/iiwa_description/iiwa7/iiwa7_no_collision.sdf) don't match the torque limits in the [iiwa documentation](https://www.kuka.com/-/media/kuka-downloads/imported/9cb8e311bfd744b4b0eab25ca883f6d3/kuka_lbr_iiwa_brochure_en.pdf).  Is there any reason I should avoid updating them?  @siyuanfeng-tri ?  @calderpg-tri ?

calderpg-tri, 2021-12-05:
See #15613 for the existing discussion on changes to the IIWA models, and #16003 for an in-progress PR on updating limits.

RussTedrake, 2021-12-05:
nice!  sorry I missed those.  I'll close this as redundant.

closed 2021-12-05 by RussTedrake