RobotLocomotion/drake #16181: <mimic> tag in urdf is silently ignored; can lead to confusing TAMSI error messages with no justification

RussTedrake, 2021-12-05:
Problem:
Someone grabs the default [franka_description](https://github.com/frankaemika/franka_ros/blob/develop/franka_description/robots/hand.xacro) from the web, and puts it in Drake.  The result is that TAMSI (reasonably) falls down, ironically at this line:
```
    // Geometry tell us that a real solution does exist i.e. Delta > 0.
    DRAKE_DEMAND(Delta > 0);
```
It doesn't exactly instill confidence, nor help someone see a path forward.

It turns out that the fingers in those models have no inertia, and are presumably relying on the `mimic` tag in the urdf http://wiki.ros.org/urdf/XML/joint, which we silently ignore.  To be honest, I'm not sure what other simulators do, because it seems like there is no inertia on either finger!  

Proposed resolution:
The real solution to this is to support equality constraints in MultibodyPlant (e.g., #14694).  In the short term, I think we should at least produce a helpful warning if we are ignoring a `mimic` tag?

sherm1, 2021-12-06:
cc @joemasterjohn @amcastro-tri constraints!

amcastro-tri, 2021-12-06:
>  I think we should at least produce a helpful warning if we are ignoring a mimic tag?

I think that is the most reasonable thing to do. Probably we should have mechanism to give a warning for any of element that Drake ignores.

jwnimmer-tri, 2021-12-15:
> Probably we should have mechanism to give a warning for any of element that Drake ignores.

Filed as #16229 now.

jwnimmer-tri, 2022-02-09:
Since #16229 has taken over the "diagnose unsupported elements during parsing" feature, should we re-purpose _this_ issue to be able the poor error message `Delta > 0` when a user forgets to add inertia to their links?  That happens quite frequently for new users, even without the particular `<mimic/>` failure mode.

amcastro-tri, 2022-02-11:
> should we re-purpose this issue to be about the poor error message Delta > 0 when a user forgets to add inertia to their links?

I believe no. The demand for `Delta > 0` can be triggered due to many reasons. It only tells you there was a problem with the numerics somewhere, but it could be anywhere.

That is, IMO we should have explicit warnings at parsing when a user forgets to add inertia to their links. I'd imagine in the future if that link with no inertia was welded to another link in the model, we could assume the link is meant to have zero mass and the user meant it to be an "extension" of the link it is welded to. IMO that's a horrible semantics, all links should have inertia specified and I would even escalate those warnings to errors.

Thoughts?

jwnimmer-tri, 2022-02-11:
That's exactly what I was trying to say -- when the user forgets to add inertia, we should fail-fast and tell them so, we should never give them the `Delta > 0` error in that specific case.  Other cases might still yell about Delta, just not the you-forgot-your-inertia case anymore.

jwnimmer-tri, 2022-02-11:
Given the discussion history here, I suppose it would be clearest to file a new issue for that missing diagnostic.

jwnimmer-tri, 2022-04-19:
I just now filed the "missing intertia" diagnostic (unrelated to the mimic tag) as #16992.  The "ignored mimic tag" issue is a duplicate of #16783, or more broadly #16229.  Therefore, I'll close this issue out in lieu of the two more-specific issues.

commit 4f4de670 (2022-04-21): [parsing] Warn about unsupported URDF features (#16953)

* [parsing] Warn about unsupported URDF features

Closes: #16783
Closes: #16181

Based on study of the ROS documentation and urdfdom module sources, add
warnings and documentation for ignored URDF features. Implementation is similar
to the in-progress mujoco parser, with an eye to future integration of
diagnostics there.

Adjust tests for existing files that emit new warnings, and add explicit
tests for each of the added warning types.

For Release Notes:
This patch modifies some installed example files to purge unsuppported
elements tha [... cut]
closed 2022-04-19 by jwnimmer-tri