RobotLocomotion/drake #16212: [bindings] Bind public APIs for hydroelastics in python.

DamrongGuoy, 2021-12-13:
To make hydroelastic contact model useful for more users, we should have python bindings for relevant public APIs including:
- drake/bindings/pydrake/geometry_py_hydro.cc
  - PolygonSurfaceMesh
  - SurfacePolygon
- drake/bindings/pydrake/geometry_py_scene_graph.cc
  - QueryObject
    - ComputeContactSurfaces()
    - ComputeContactSurfacesWithFallback()
  - HydroelasticContactRepresentation
  - ContactSurface
      - poly_mesh_W()
      - is_triangle()

Relates #15796. 

I'm not sure yet how extensive the bindings should be.  It might depend on the python examples that we want to create.

@hongkai-dai, please let us know if you have a list of specific bindings that you'd like to see. You have more experience as python users of hydroelastics.

@amcastro-tri, I only list the items from `geometry`. Please let us know if there are items from `multibody` too (you can add to the list above or add a comment below).

jwnimmer-tri, 2021-12-14:
@DamrongGuoy is there any API that is still planned to be changed under the #15796 roadmap?  From a quick skim of the issue, I don't see any.

DamrongGuoy, 2021-12-14:
@jwnimmer-tri Thanks for asking.  Most of the API should stay now with two variables:
- #16218 
  - In the PR, I assume the reviewers will let me keep the name AddSoftHydroelasticProperties(), so there won't be changes in API.
 - @DamrongGuoy will extend the new API to let users choose between triangles and polygons and update MultibodyPlantConfig as needed. However, this is not a requirement for the official release.
   - The PR will hopefully start in a few days.

jwnimmer-tri, 2021-12-14:
> Should we bind MeshFieldLinear ...

I'll presume "no", for now.

jwnimmer-tri, 2021-12-14:
In general I think that every public (non-internal, non-dev, etc.) class and function from C++ should be bound in pydrake.  Adding the binding only when people complain is an awkward development philosophy.  However, for the purpose of this issue and the #15796 milestone, I think the current pull request is sufficient to close this issue, even without adding more bindings.  Part of the post-milestone follow-up should be ensuring that all of multibody and geometry code is bound in pydrake.

PR #16219 (2021-12-14): [pydrake] Bind some hydroelastic-related classes and functions
Closes #16212.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16219)
closed 2021-12-14 by jwnimmer-tri