RobotLocomotion/drake #16215: Deprecation of drake-visualizer

jwnimmer-tri, 2021-12-14:
As our supported platforms continue to evolve (adding Ubuntu 22 per #13391 or adding Apple ARM support per #13514), it is becoming increasingly more difficult to keep patching the Director sources to operate correctly, given the base system churn.

At some point, I think Drake will need to withdraw support for `drake-visualizer`.  This issue exists to discuss that idea, and put a specific action plan into place.

My current thought is that we should freeze support for `drake-visualizer` at the current set of platforms (Ubuntu 18 Bionic, Ubuntu 20 Focal, macOS 11 Big Sur x86_64, macOS 12 Monterey x86_64).  When Drake adds support for a new platform (Ubuntu 22, macOS 13, macOS ARM), we will not include `drake-visualizer` in that support footprint.

Per our "two Ubuntu" policy and #13391, we will drop support for Ubuntu 18 next April 2022; therefore, that `drake-visuzlier` build flavor would also be dropped.  Given the relative stability of the Ubuntu platform, it seems plausible to continue to support our `drake-visualizer` build on Ubuntu 20 for as long as Drake supports Ubuntu 20 (estimated to last through early 2024).  Per #14692, we might distribute it via a different release channel and cadence, but it would still exist as part of the overall project.

Per our "two macOS" policy, overall support for Big Sur would be dropped when the next macOS (13?) lands, anticipated in October 2022.  At most, that means the Monterey build of `drake-visualizer` could last until fall of 2023.  However, macOS support of the visualizer is one of the most challenging problems we face (given the evergreen, non-stability promises of Homebrew).  I think it's eating up a lot of our developer hours, with relatively small gain.  I believe that many of our macOS users focus on pydrake, meshcat, and jupyter, and so don't necessarily use `drake-visualizer`.  Therefore, my proposal is to deprecate `drake-visualizer` on macOS as of today, and remove it from Drake as of April 1, 2022 (concurrently with dropping Bionic).

In the meantime, we should work to port all of our demos to use the meshcat C++ API (or its pydrake bindings).  In particular, some of the complex hydroelastic mesh/patch visualizations will need some investigation for how to port to use meshcat efficiently.

Also worth noting is that if any Drake community member would like to take over maintenance of `drake-visualizer` past the support windows I propose above, that is another welcome option.  We could keep it as another repository within the RobotLocomotion organization, offer CI machine resources, etc.

To summarize, here is the specific support matrix / timeline for `drake-visualizer` that I'm proposing, as a starting point:
- Ubuntu 18 Bionic: through April 1, 2022 (same as the rest of Drake's support for this platform).
- Ubuntu 20 Focal: through early 2024 (same as the rest of Drake's support for this platform).
- macOS 11 Big Sur x86_64: through April 1, 2022 (EOL prior to the rest of Drake's support for this platform).
- macOS 12 Monterey x86_64: through April 1, 2022 (EOL prior to the rest of Drake's support for this platform).
- macOS on ARM: never supported.

jwnimmer-tri, 2021-12-14:
Tasks:
- [ ] #16216
- [ ] More TBD

jwnimmer-tri, 2021-12-15:
It's also worth noting that even though the work above is phrased in terms of `meshcat` as primary visualization, another option in the future will be https://github.com/RobotLocomotion/drake-ros which will connect Drake to RViz.

jwnimmer-tri, 2022-01-20:
For the record, our plan is to use "meshcat" as the preferred visualizer.  We'll have more demos of that up shortly (#16216).

jwnimmer-tri, 2022-04-13:
In #16944, we dropped macOS support.  Remaining porting work of demos to Meshcat is tracked under #16216.

closed 2022-04-13 by jwnimmer-tri

richardrl, 2023-10-06:
As I mentioned to @jwnimmer-tri , drake-visualizer serves a unique purpose right now for certain applications because it is much, much faster than meshcat (10x). For example, for my VR teleop demo, it was impossible to get real-time rates with meshcat. I would hesitate dropping support for the drake-visualizer binary for this reason. 

@liruiw also has experienced the slowness

jwnimmer-tri, 2023-11-06:
> I would hesitate dropping support for the drake-visualizer binary for this reason.

Drake continues to support the `drake-visualizer` LCM protocol.  So the only question is whether you have access to a `drake-visualizer` binary to run.  Drake no longer ships such a binary for Jammy or macOS, and will stop shipping the Focal binaries next spring.  You are welcome to still keep using any previously-shipped binaries that still work for you.

All of the code for `drake-visualizer` is open-source at https://github.com/RobotLocomotion/director along with some [Drake-specific patches](https://github.com/RobotLocomotion/drake/tree/v1.22.0/tools/workspace/drake_visualizer/patches) and [Drake-specific plugins](https://github.com/RobotLocomotion/drake/tree/v1.22.0/tools/workspace/drake_visualizer/_drake_visualizer_builtin_scripts) which are also open source. You have the power to fork the `director` repository, incorporate any patches or plug-ins that you care about, and build your own binaries that meet your needs.

TRI does not have sufficient resources to maintain the abandoned https://github.com/RobotLocomotion/director project indefinitely.  We did so for a few years but in early 2024 we've drawn a line in the sand.  People who still want to use it are invited to fork it and cooperate to maintain it for their own needs.

> ... it was impossible to get real-time rates with meshcat ...

We would welcome any pull requests that help to speed up Meshcat, and would help to get them merged into Drake and/or Meshcat.