RobotLocomotion/drake #16216: Port examples away from drake-visualizer

jwnimmer-tri, 2021-12-14:
We should stop using `drake-vizualizer` in our examples (see #16215).  This issue maintains the checklist of programs that we'll need to port to use meshcat (or otherwise).

- [ ] TBD

SeanCurtis-TRI, 2021-12-14:
Worth noting that *anything* using `ContactResultsToLcm` to visualize hydroleastic contact has no `drake_visualizer`-free substitute. Adding this missing functionality is one of the prereqs.

xuchenhan-tri, 2021-12-14:
I'll take on one (the example I introduced) as an experiment. I may take on more in the future as I get more working knowledge of MeshCat.

jwnimmer-tri, 2021-12-20:
#16248 actually raises an interesting policy question that I failed to consider during review.  We _could_ have both visualizer options enabled for demos like these, even if we document / steer people towards meshcat as the preferred one.  In this example at least, it's only an extra line of code for each one.

xuchenhan-tri, 2021-12-20:
What's the point of keeping the drake_visualizer option if we are not going to deprecate it in the coming months?

jwnimmer-tri, 2021-12-20:
The current plan in #16215 says that for Ubuntu 20 (only), we'd continue to support `drake-visualizer` through early 2024.  For that, we should probably preserve at least _some_ examples that allow for demonstrating and regression testing it.  I suppose doing it for all of them would be pointless (or even misleading), though.

Maybe we nominate the `examples/kuka_iiwa_arm` as the one that should preserve `drake-visualizer` support, and everything else swaps to be meshcat-only.

jwnimmer-tri, 2023-07-29:
This will be mostly finished as of #19781.  Instead of spending time porting all of the demos to Meshcat, we'll just switch them to use Meldis and call it a day.

There are a few hydroelastic demos that still seem to require `drake_visualizer`.  Those will be tracked in #19870.

PR #19871 (2023-08-01): Clearly document the drake_visualizer program as legacy code
- Anywhere we mention `drake_visualizer` refer to it as "legacy ... of days past".
- Replace ipynb mentions of `drake_visualizer` with Meldis.
- Remove visualization from realsense parsing test; that's what `//tools:model_visualizer` is for.

Closes #16216.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19871)
closed 2023-08-01 by rpoyner-tri