RobotLocomotion/drake #16217: Ubuntu 22.04 (Jammy Jellyfish) support

jwnimmer-tri, 2021-12-14:
This is an anchor issue for discussing goals and timelines for 22.04 support.

My starting bid for Drake's support timeline would be 22.04 release date (scheduled for late April 2022) + 60 days for having nighty binary releases ready. The mid-July Drake release would be the first candidate for supporting 22.04.

For reference, the Ubuntu 20 support ticket was #13102, and dropping Ubuntu 18 ticket is #13391.

jwnimmer-tri, 2022-04-13:
For the "for drake maintainers only" developer tools (doxygen, sphinx, release notes, dependency upgrades, etc.), we should _not_ endeavor to add Jammy support.  This scope of this issue is only the user-facing stuff (library code, tests, demos, tutorials, etc.).

jwnimmer-tri, 2022-04-13:
For the same reason, we can also defer the memory checker builds (sanitizer, valgrind, etc) for the moment.  The priority would be the packaging builds and their nearby support, e.g., Release, Debug, and Everything.

svenevs, 2022-04-13:
Related bionic deprecations as we prepare for this: https://github.com/RobotLocomotion/drake/issues/13391#issuecomment-1097043824

- **No** `drake_visualizer` support for Jammy will be added.  #16215 
- [ ] `tools/workspace/vtk` binaries will be added.
- [ ] `tools/workspace/intel_realsense_ros`: re-pin to the matching noetic baseline
- [ ] `tools/workspace/pygame_py`: switch to ubuntu `python3-pygame` package
- [ ] `tools/workspace/ros_xacro`: re-pin to the matching noetic baseline

jwnimmer-tri, 2022-04-14:
I think the work plan here would be something like:
- [ ] Add definitions of Unprovisioned Experimental CI builds for Jammy (Release, Debug, Everything).
  - No sanitizers, no wheels, no documentation, no packaging, etc.
- [ ] Develop a pull request that adds Jammy support to Bazel Release builds (workspace, os.bzl, etc).
- [ ] Add Ubuntu 22 AMIs.
- [ ] Add definitions of Provisioned Experimental CI builds for Jammy (of the same flavors).
- [ ] Incrementally work on standing up Debug, Everything, etc. builds in Continuous (non-Production) and/or Nightly.
- [ ] Add Packaging builds.
- [ ] Once that's all working, announce official support, move into Production, and add to buildcop.

jwnimmer-tri, 2022-06-08:
I fired off a current [Jammy canary build](https://drake-jenkins.csail.mit.edu/view/Linux%20Jammy%20Unprovisioned/job/linux-jammy-unprovisioned-gcc-bazel-experimental-release/2/console).  Looks like some `-Werror=range-loop-construct` problems that could be a nice first tidy-up, and some missing includes in `#ibex` where we'll probably need a local patch file to add them back.   I didn't look more carefully than that yet.

svenevs, 2022-06-08:
@mwoehlke-kitware has more info on the `#include <numeric_limits>` ibex issue. My recollection is it's a multi faceted issue.

jwnimmer-tri, 2022-06-08:
Great! I'm happy to have you all triage and own the work.  If there's a particularly troublesome area where collaborating with someone who has experience with a piece of code would help, feel free to ask on slack, etc.

mwoehlke-kitware, 2022-06-08:
Ibex is hopefully straight-forward, though this might be a good opportunity to look into moving back to upstream?

The massive headache is clang-cindex-python3. There's an update available (see also #17315), but updating causes other stuff to break in a way that's not at all obvious. I'm probably going to need help with that.

jwnimmer-tri, 2022-06-08:
> Ibex... opportunity to look into moving back to upstream?

Fine by me.

I'll look at the clang thing (tomorrow?).

jwnimmer-tri, 2022-06-09:
Goal to have this passing: ~3-4 weeks.

mwoehlke-kitware, 2022-06-21:
I've also filed #17433 as I've not been able to make any progress on finding the cause of those test failures. It appears that and websockets (https://github.com/RobotLocomotion/drake/issues/16217#issuecomment-1162252617) may be the only remaining issues, at least as far as getting CI to pass.

jwnimmer-tri, 2022-06-21:
> At least meshcat is blocked by https://bugs.launchpad.net/ubuntu/+source/python-websockets/+bug/1969902.

On first glance, it seems like it might be possible to monkey-patch this in the one unit test that uses it.  Maybe extend `WebSocketClientProtocol` and have the subclass constructor discard the loop variable when calling super?

mwoehlke-kitware, 2022-06-22:
> On first glance, it seems like it might be possible to monkey-patch this in the one unit test that uses it.

Not AFAICT, and if a work-around were possible, I'd expect that to be noted in the Ubuntu bug report. The problem is in `/usr/lib/python3/dist-packages/websockets/legacy/protocol.py`, which *always* passes a `loop` parameter. It ***already*** passes `loop=None` (so "discard[ing] the loop variable" won't have any effect). That entire ctor would need to be replicated at the least, and it's not a trivial chunk of code. I'd be fairly paranoid about trying to monkey-patch it.

If I was going to monkey-patch anything, it would be `asyncio`, and it would have to be an *actual* monkey-patch.

I am hopeful that this will get fixed. AFAICT, python3-websockets is *completely unusable* in Jammy, and this doesn't seem like it would be so much a fringe package that it will just be left broken for Jammy's entire life.

jwnimmer-tri, 2022-06-22:
> I am hopeful that this will get fixed.

Me too, but will it ship to Ubuntu 22 within the next 2 weeks?  That's the deadline for this ticket, so that we can add Jammy support in our mid-July monthly release.

---

Does something like this patch fix it on Jammy?

```patch
--- a/geometry/test/meshcat_websocket_client.py
+++ b/geometry/test/meshcat_websocket_client.py
@@ -42,9 +42,19 @@ def print_recursive_comparison(d1, d2, level='root'):
             print(f"{level:<20} {repr(d1)} != {repr(d2)}")
 
 
+class MyWebSocketClientProtocol(websockets.WebSocketClientProtocol):
+    """Work-around for websockets bug on Ubuntu 22.04.
+    https://bugs.launchpad.net/ubuntu/+source/python-websockets/+bug/1969902"""
+    def __init__(self, **kwargs):
+        kwargs.pop("loop")
+        super().__init__(**kwargs)
+
+
 async def socket_operations_async(args):
     logger.info("Connecting...")
-    async with websockets.connect(args.ws_url, timeout=1) as websocket:
+    async with websockets.connect(
+            args.ws_url, timeout=1,
+            klass=MyWebSocketClientProtocol) as websocket:
         logger.info("... connected")
         if args.send_message is not None:
             logger.info("Sending...")
```

mwoehlke-kitware, 2022-06-22:
No, it doesn't help, for the reasons explained in my previous comment.

I tried this:
```python
_asyncio_lock_ctor = asyncio.Lock.__init__
_asyncio_wait_for = asyncio.wait_for
_asyncio_sleep = asyncio.sleep

def _patch_asyncio(orig):
    def _patched(*args, **kwargs):
        if 'loop' in kwargs and kwargs['loop'] is None:
            kwargs.pop('loop')
        return orig(*args, **kwargs)

    return _patched

asyncio.Lock.__init__ = _patch_asyncio(_asyncio_lock_ctor)
asyncio.wait_for = _patch_asyncio(_asyncio_wait_for)
asyncio.sleep = _patch_asyncio(_asyncio_sleep)```

That seems to help, but I still get:
```
Traceback (most recent call last):
  File "/root/.cache/bazel/_bazel_root/9ecb0f4ce434715009a64c73d901046c/sandbox/processwrapper-sandbox/78/execroot/drake/bazel-out/k8-opt/bin/geometry/meshcat_test.runfiles/drake/geometry/test/meshcat_websocket_client.py", line 126, in <module>
    main()
  File "/root/.cache/bazel/_bazel_root/9ecb0f4ce434715009a64c73d901046c/sandbox/processwrapper-sandbox/78/execroot/drake/bazel-out/k8-opt/bin/geometry/meshcat_test.runfiles/drake/geometry/test/meshcat_websocket_client.py", line 112, in main
    loop.run_until_complete(
  File "/usr/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/root/.cache/bazel/_bazel_root/9ecb0f4ce434715009a64c73d901046c/sandbox/processwrapper-sandbox/78/execroot/drake/bazel-out/k8-opt/bin/geometry/meshcat_test.runfiles/drake/geometry/test/meshcat_websocket_client.py", line 77, in socket_operations_async
    parsed = umsgpack.unpackb(message)
  File "/usr/lib/python3/dist-packages/umsgpack.py", line 867, in _unpackb3
    return _unpack(io.BytesIO(s), options)
  File "/usr/lib/python3/dist-packages/umsgpack.py", line 693, in _unpack
    return _unpack_dispatch_table[code](code, fp, options)
  File "/usr/lib/python3/dist-packages/umsgpack.py", line 677, in _unpack_map
    elif not isinstance(k, collections.Hashable):
AttributeError: module 'collections' has no attribute 'Hashable'
```

Maybe we should just vendor [u-msgpack-python](https://github.com/vsergeev/u-msgpack-python) and [websockets](https://github.com/aaugustin/websockets)? Or is there a way to use the latest PyPI versions rather than the old cruft Jammy is shipping via apt?

mwoehlke-kitware, 2022-06-22:
Note: https://bugs.launchpad.net/ubuntu/+source/u-msgpack-python/+bug/1979549

jwnimmer-tri, 2022-06-22:
For `u-msgpack-python` we could do `@umsgpack_py_internal` in the workspace (akin to `@uritemplate_py_internal`) since there is no native code.  Maybe that, plus the monkey-patch is sufficient?

For `websockets` which has native code, we do have https://github.com/RobotLocomotion/drake/blob/master/tools/workspace/pypi_wheel.bzl to help but it might end up being wonky to get everything loading correctly.

jwnimmer-tri, 2022-06-23:
The "Nightly" part looks good.  Let's hold off on any "Continuous" jammy jobs until we have provisioned images ready (in ~3 weeks or whatever).  I don't really want to burn the $$ on long, uncached continuous jobs yet.

BetsyMcPhail, 2022-06-24:
The PR is ready to go when we're ready for the nightly jobs. Note that it's not enough to merge the PR, `jenkins-jobs` must be run with the updated `jobs.yaml`.

jwnimmer-tri, 2022-06-30:
CI builds using CMake [sayeth](https://drake-jenkins.csail.mit.edu/view/Linux%20Jammy%20Unprovisioned/job/linux-jammy-unprovisioned-gcc-cmake-experimental-release/1/):
```
[2022-06-30T14:12:44.744Z] CMake Error at CMakeLists.txt:85 (message):
[2022-06-30T14:12:44.744Z] Release jammy is NOT supported.  Please use Ubuntu 20.04 (Focal).
```

I think we are still missing a Drake PR to remove that fast-fast @mwoehlke-kitware.

mwoehlke-kitware, 2022-06-30:
> I think we are still missing a Drake PR to remove that fast-fast

(Typo for "fail-fast"?) See #17480.

jwnimmer-tri, 2022-07-01:
@mwoehlke-kitware the Jammy Debug builds [are failing](https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=1613428).  The IBEX `<numeric_limits>` patch needs to be exteneded to cover uses within `assert(...)` as well.

BetsyMcPhail, 2022-07-05:
See #17493 for IBEX debug issues

jwnimmer-tri, 2022-07-05:
The Jammy Experimental builds are all green now.  I think we're ready to add and enable the Jammy Nightly now.

BetsyMcPhail, 2022-07-05:
Nightly Jammy jobs (listed above) are in Jenkins and should run tonight.

BetsyMcPhail, 2022-07-07:
See [drake-ci PR #169](https://github.com/RobotLocomotion/drake-ci/pull/169) for enabling remote caching.

jwnimmer-tri, 2022-07-07:
Re: Wheel -- sounds good.  It would actually be _wrong_ to add a "Jammy Wheel" build to Jenkins.  The Python 3.10 wheels for Jammy users will be compiled on Focal (in our Dockerfile stuff).

jwnimmer-tri, 2022-07-08:
I launched the Experimental Packaging job [here](https://drake-jenkins.csail.mit.edu/view/Linux%20Jammy%20Unprovisioned/job/linux-jammy-unprovisioned-gcc-bazel-experimental-snopt-mosek-packaging/1/console).  It made it almost to the end (built tgz ok, uploaded tgz ok), but then it hit this:

```
[5:44:59 PM]  'sudo' '/usr/bin/docker' 'build' '-t' 'robotlocomotion/drake:jammy' '-t' 'robotlocomotion/drake:jammy-20220708' '.'
[5:44:59 PM]  unable to prepare context: unable to evaluate symlinks in Dockerfile path: lstat /media/ephemeral0/ubuntu/workspace/linux-jammy-unprovisioned-gcc-bazel-experimental-snopt-mosek-packaging/src/tools/install/dockerhub/Dockerfile: no such file or directory
```

It looks like https://github.com/RobotLocomotion/drake/tree/master/tools/install/dockerhub/ needs a PR to add jammy support.  Can you take a look?

Once that PR lands, please add a Nightly flavor of that same job, and elevating to Production when/once it's green.

BetsyMcPhail, 2022-07-11:
Checklist to help keep myself organized:

- [x] Add "experimental" full matrix to match Focal; they will not be pre-merge so will never trigger unless we ask manually.
- [x] Add Jammy caching
- [ ] Add one "continuous provisioned" linux-jammy-gcc-bazel-continuous-release (production)
- [ ] Add tgz packaging build (to be run nightly, _not_ Production).
- [ ] Elevate tgz packaging to Production
- [ ] Add weekly provisioned sanitizers (_not_ Production).

mwoehlke-kitware, 2022-07-13:
FYI: https://github.com/RobotLocomotion/drake/issues/17562 (tsan issue on Jammy)

jwnimmer-tri, 2022-07-13:
To be clear, the current goal for Jammy is to _run_ the dynamic analysis builds so that we have their information available, but to leave them broken.  There is no urgency to making Jammy perfect.  Everything else already on the project board is more important.

jwnimmer-tri, 2022-07-14:
If there are _passing_ Sanitizer builds, it's fine to move them into Weekly Production just so we get an email in case they go red.

svenevs, 2022-07-14:
Jammy nightly production should kick off tonight, sanitizers this weekend.  We will be promoting whatever is passing on Monday to production :+1:

BetsyMcPhail, 2022-07-19:
The Jammy nightly packaging job has been moved to production --> https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-unprovisioned-gcc-bazel-nightly-snopt-mosek-packaging/

[drake-jenkins-job PR#22](https://github.com/RobotLocomotion/drake-jenkins-jobs/pull/22)

BetsyMcPhail, 2022-07-19:
The subset of passing Jammy weekly sanitizer jobs has been moved to production. There are 4 remaining Jammy weekly *san non-production jobs that are **not** expected to pass.

[drake-jenkins-jobs PR#23](https://github.com/RobotLocomotion/drake-jenkins-jobs/pull/23)

BetsyMcPhail, 2022-07-19:
With that, I believe we can call this issue closed.

closed 2022-07-19 by BetsyMcPhail