RobotLocomotion/drake #16244: [multibody] Hydroelastics with polygon contact surfaces crash continuous systems.

DamrongGuoy, 2021-12-17:
After PR #16238 was merged, users of hydroelastic contact models can freely mix-and-match {continuous, discrete} system with {triangle,polygon} contact surfaces by using this switch in MultibodyPlant:
```C++
    // Use polygon contact surfaces.
    plant.set_contact_surface_representation(
        geometry::HydroelasticContactRepresentation::kPolygon);
    // Use triangle contact surfaces.
    plant.set_contact_surface_representation(
        geometry::HydroelasticContactRepresentation::kTriangle);
```
I found a simple example that when I used a continuous system with hydroelastics and kPolygon, it crashed with this error message:
```
Encountered singular articulated body hinge inertia for body node index 1. Please ensure that this body has non-zero inertia along all axes of motion.
```

<h2>Reproduce: Continuous System Crash</h2>
Apply this patch on drake/examples/multibody/rolling_sphere/rolling_sphere_run_dynamics.cc to use polygon contact surfaces.

```
diff --git a/examples/multibody/rolling_sphere/rolling_sphere_run_dynamics.cc b/examples/multibody/rolling_sphere/rolling_sphere_run_dynamics.cc
index 7fb9773d4..151a35495 100644
--- a/examples/multibody/rolling_sphere/rolling_sphere_run_dynamics.cc
+++ b/examples/multibody/rolling_sphere/rolling_sphere_run_dynamics.cc
@@ -138,6 +138,8 @@ int do_main() {
   // Set contact model and parameters.
   if (FLAGS_contact_model == "hydroelastic") {
     plant.set_contact_model(ContactModel::kHydroelastic);
+    plant.set_contact_surface_representation(
+        geometry::HydroelasticContactRepresentation::kPolygon);
     plant.Finalize();
   } else if (FLAGS_contact_model == "point") {
     // Plant must be finalized before setting the penetration allowance.
```

Then, run this command to get the error:

```
(master)*$ bazel run //examples/multibody/rolling_sphere:rolling_sphere_run_dynamics -- --contact_model=hydroelastic --mbp_dt=0.0
INFO: Analyzed target //examples/multibody/rolling_sphere:rolling_sphere_run_dynamics (0 packages loaded, 0 targets configured).
INFO: Found 1 target...
Target //examples/multibody/rolling_sphere:rolling_sphere_run_dynamics up-to-date:
  bazel-bin/examples/multibody/rolling_sphere/rolling_sphere_run_dynamics
INFO: Elapsed time: 0.646s, Critical Path: 0.00s
INFO: 1 process: 1 internal.
INFO: Build completed successfully, 1 total action
INFO: Build completed successfully, 1 total action
terminate called after throwing an instance of 'std::runtime_error'
  what():  Encountered singular articulated body hinge inertia for body node index 1. Please ensure that this body has non-zero inertia along all axes of motion.
Aborted (core dumped)
```

<h2> Verify: Discrete System OK </h2>
By changing `mbp_dt` to non-zero (1e-3 seconds), we will get a discrete system that runs fine like this:

```
(master)*$ bazel run //examples/multibody/rolling_sphere:rolling_sphere_run_dynamics -- --contact_model=hydroelastic --mbp_dt=1e-3
INFO: Analyzed target //examples/multibody/rolling_sphere:rolling_sphere_run_dynamics (0 packages loaded, 0 targets configured).
INFO: Found 1 target...
Target //examples/multibody/rolling_sphere:rolling_sphere_run_dynamics up-to-date:
  bazel-bin/examples/multibody/rolling_sphere/rolling_sphere_run_dynamics
INFO: Elapsed time: 0.387s, Critical Path: 0.00s
INFO: 1 process: 1 internal.
INFO: Build completed successfully, 1 total action
INFO: Build completed successfully, 1 total action
Simulator::AdvanceTo() wall clock time: 9.995 seconds.
General stats regarding discrete updates:
Number of time steps taken (simulator stats) = 2112
Simulator publishes every time step: false
Number of publishes = 129
Number of discrete updates = 2000
Number of "unrestricted" updates = 0

Note: the following integrator took zero steps. The simulator exclusively used the discrete solver.

Stats for integrator RungeKutta3Integrator with error control:
Number of time steps taken (integrator stats) = 0
Initial time step taken = nan        s
Largest time step taken = nan        s
Smallest adapted step size = nan        s
Number of steps shrunk due to error control = 0
Number of derivative evaluations = 0
Number of steps shrunk due to convergence-based failure = 0
Number of convergence-based step failures (should match) = 0
```

DamrongGuoy, 2021-12-17:
CC: @rpoyner-tri @amcastro-tri @sherm1

amcastro-tri, 2021-12-18:
The error above was a clear indication we had NANs somewhere, especially for such simple case with just a sphere.
Following the trace of NANs took me to a bug in the computation of centroids.
Fixed in #16253, PTAL. I believe worth introducing the same fix for computations involving mesh of triangles as well.

DamrongGuoy, 2021-12-19:
Thank you for the quick diagnostic.  I do not see the word `NAN` in `Encountered singular articulated body hinge inertia for body node index 1. Please ensure that this body has non-zero inertia along all axes of motion,` but I believe what you said.

It sounds like sometimes we got a zero-area or near-zero-area polygon in the plane vs. tetrahedral mesh intersection code, or the halfspace vs. triangle mesh intersection code.  That sounds reasonable because I vaguely remember the numerical threshold in those codes was very very strict.

On the other hand, I believe we already put a reasonable numerical threshold for the triangle mesh vs. tetrahedral mesh intersection code. I have not been able to reproduce a similar problem for that code.

What worries me about #16253 is that we are punishing a good-citizen code (mesh-mesh intersection with reasonable numerical threshold) due to problems in other parties (plane/tet-mesh or halfspace/tri-mesh intersection code).  The "robust" centroid computation in #16253 worries me that it taxes every polygon, good one or bad one.

Let me do simple thresholding in the plane/tet-mesh and halfspace/tri-mesh code.  If it fixes this example, I'll close this issue.  Otherwise, I'll study #16253 again to see whether we can make it robust, fast, and simple at the same time.

Again thank you for analyzing the problems.

Change Assignee to @DamrongGuoy .

amcastro-tri, 2021-12-19:
> Let me do simple thresholding in the plane/tet-mesh and halfspace/tri-mesh code. If it fixes this example, I'll close this issue.

I know the answer: it will fix it. The problem with such a fix is that we won't be able to tell if it affect the numerics of our solvers. Thresholding usually leads to on/off in the numerics that causes very difficult to track problems. That is, things like integrators could have a harder time. Impossible to tell without trying it out, the kind of thing will go undetected for years until someone pays close attention to it.

I can see however how thresholding is simpler though, it will definitely solve this one problem. The fix to #16253 is trivial if you want me to use such an approach.

amcastro-tri, 2021-12-19:
In #16255 I implemented a solution more in line with what we have in master. I think you'll like that better than #16253, @DamrongGuoy.

DamrongGuoy, 2021-12-20:
Thank you @amcastro-tri .  We both work on Sunday because we care so much about hydroelastics.

I think we're converging now. Your #16255 is very similar to my #16257 + #16258.

Let's talk face to face in our coming meetings. I believe we're near the agreed solution now.

amcastro-tri, 2021-12-20:
@DamrongGuoy, thanks for looking into this. Just notice that you assigned this to me with high priority. In the future please unassign others from an issue if you plan to work on it yourself to avoid duplicating work.

DamrongGuoy, 2021-12-21:
I close #16253 and #16255 since we'll replace them with #16257 + #16258.

@amcastro-tri  Sorry for the confusion.  At first, I set high priority because I didn't know how difficult/easy to get from that error message to the root cause (zero-area polygon).  Once you gave the insight of what's the root cause, I should have set it to low priority but forgot. I did assign the issue back to me when I know it's as simple as zero-area polygons, not something else that is more serious.

Instead of work duplication, I see it as evaluating several options in real-time. :smile:   In the end, I'll combine both your ideas and my ideas into #16257 + #16258 (still drafts for both PRs).

DamrongGuoy, 2022-01-05:
#16257 is ready for review now.  Once it's merged, we can start #16258.

DamrongGuoy, 2022-01-06:
Both #16257 and #16258 are merged into master now. I close this issue.

closed 2022-01-06 by DamrongGuoy