RobotLocomotion/drake #16262: pydrake yaml_dump() of a schema object fails

ggould-tri, 2021-12-20:
The python expression `yaml_dump(UniformVectorX(min=[1.], max=[2.]))` surprisingly fails.
Per https://app.slack.com/client/T0JNB2DS4/C3YB3EV5W/thread/C3YB3EV5W-1640023403.047200 this is the intended behaviour (yaml_dump is not intended to be class-aware) but from a usability perspective it's a bit obnoxious because, correct construction of the tagged map for a schema object is typo-prone and can cause confusing runtime errors.

sherm1, 2021-12-22:
Assigning to @ggould-tri initially -- please feel free to reassign.

jwnimmer-tri, 2023-06-14:
The missing feature we want is #19597.  After that, the call to action for this ticket is to make sure that the `yaml_dump` example above gives a good message that points people to `yaml_dump_typed`.

jwnimmer-tri, 2024-09-17:
... and that is finished.  The only remaining work here is to check and improve the error message.

PR #21935 (2024-09-23): [py common] Improve a yaml_dump error message
Closes #16262.  I was in the neighborhood, so knocked this out quickly.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/21935)
closed 2024-09-23 by jwnimmer-tri