RobotLocomotion/drake #16292: symbolic::Polynomial::DoAddProduct spends way too much time in Expand()

hongkai-dai, 2021-12-30:
I have a dev branch that does a lot of symbolic::Polynomial operations. The code is unnecessarily slow. When I run callgrind, I found that the program spends a lot of time on this block within `DoAddProduct` function
https://github.com/RobotLocomotion/drake/blob/2ad76f8ca579bc2883faf6d7331cae4bb57256fe/common/symbolic_polynomial.cc#L46-L50
Particularly calling `Expand()` function takes more than 70% of time in my program. Here is a screen shot of kcachegrind, note that most of the time is spend on DoAddProduct

[image]

I think this block is inefficient, for several reasons
1. We evaluate the sum of the expanded expression `existing_coeff.Expand() + coeff.Expand()`, and later again do `existing_coeff += coeff`.
2. After we do `existing_coeff += coeff`, will this `existing_coeff` be an expanded expression? I guess the answer is No. If we have the sum of three polynomials `p1(x) + p2(x) + p3(x)`, then when we evaluate the sum of `p1(x) + p2(x)`, we will expand their coefficient expressions, and then when we add `p3(x)` to the sum `p1(x) + p2(x)`, we will need to expand the coefficient expressions in `p1(x) + p2(x)` again.

I made the following change to this code block to save the result of `existing_coeff.Expand() + coeff.Expand()`
```
diff --git a/common/symbolic_polynomial.cc b/common/symbolic_polynomial.cc
index aae335f47b..67ed6caeaf 100644
--- a/common/symbolic_polynomial.cc
+++ b/common/symbolic_polynomial.cc
@@ -43,10 +43,11 @@ void DoAddProduct(const Expression& coeff, const Monomial& m,
     // Without expanding the terms, we have `is_zero(c1 + c2) = false` while
     // it's clear that c1 + c2 is a zero polynomial. Using `Expand()` help us
     // identify those cases.
-    if (is_zero(existing_coeff.Expand() + coeff.Expand())) {
+    const symbolic::Expression coeff_sum = existing_coeff.Expand() + coeff.Expand();
+    if (is_zero(coeff_sum)) {
       map->erase(it);
     } else {
-      existing_coeff += coeff;
+      existing_coeff = coeff_sum;
     }
   } else {
     // m ∉ dom(map)
```

Now the total runtime of my program drops from 24s to 15s.

If I further completely remove the call to `Expand()` with this patch
```
diff --git a/common/symbolic_polynomial.cc b/common/symbolic_polynomial.cc
index aae335f47b..7bfc4c925b 100644
--- a/common/symbolic_polynomial.cc
+++ b/common/symbolic_polynomial.cc
@@ -43,11 +43,11 @@ void DoAddProduct(const Expression& coeff, const Monomial& m,
     // Without expanding the terms, we have `is_zero(c1 + c2) = false` while
     // it's clear that c1 + c2 is a zero polynomial. Using `Expand()` help us
     // identify those cases.
-    if (is_zero(existing_coeff.Expand() + coeff.Expand())) {
-      map->erase(it);
-    } else {
+    //if (is_zero(existing_coeff.Expand() + coeff.Expand())) {
+    //  map->erase(it);
+    //} else {
       existing_coeff += coeff;
-    }
+    //}
   } else {
     // m ∉ dom(map)
     map->emplace_hint(it, m, coeff);
```
Then my runtime further drops to 9s.

So I would propose two ideas
1. Cache the result of `existing_coeff.Expand() + coeff.Expand()` as in my first patch, this is an easy fix.
2. Provide an option to `DoAddProduct` and other functions in symbolic::Polynomial class, that the user can opt not to erase the term when the coefficient of some monomial is 0. This would avoid calling `Expand()` function.

@soonho-tri what do you think? I can file the PR with my first patch pretty easily. Maybe we can discuss it after the holidays.

@AlexandreAmice this is one of the remaining speed bottleneck in our SOS work.

hongkai-dai, 2022-03-24:
I created a benchmark test branch https://github.com/hongkai-dai/drake/tree/benchmark_polynomial_sum. Here is the benchmark result

1. Drake master takes 30 ms
2. By using the patch
    ```
    diff --git a/common/symbolic_polynomial.cc b/common/symbolic_polynomial.cc
    index aae335f47b..67ed6caeaf 100644
    --- a/common/symbolic_polynomial.cc
    +++ b/common/symbolic_polynomial.cc
    @@ -43,10 +43,11 @@ void DoAddProduct(const Expression& coeff, const Monomial& m,
         // Without expanding the terms, we have `is_zero(c1 + c2) = false` while
         // it's clear that c1 + c2 is a zero polynomial. Using `Expand()` help us
         // identify those cases.
    -    if (is_zero(existing_coeff.Expand() + coeff.Expand())) {
    +    const symbolic::Expression coeff_sum = existing_coeff.Expand() + coeff.Expand();
    +    if (is_zero(coeff_sum)) {
           map->erase(it);
         } else {
    -      existing_coeff += coeff;
    +      existing_coeff = coeff_sum;
         }
       } else {
         // m ∉ dom(map)
    ```
    takes 25ms
3. By using the patch
    ```
    diff --git a/common/symbolic_polynomial.cc b/common/symbolic_polynomial.cc
    index aae335f47b..7bfc4c925b 100644
    --- a/common/symbolic_polynomial.cc
    +++ b/common/symbolic_polynomial.cc
    @@ -43,11 +43,11 @@ void DoAddProduct(const Expression& coeff, const Monomial& m,
         // Without expanding the terms, we have `is_zero(c1 + c2) = false` while
         // it's clear that c1 + c2 is a zero polynomial. Using `Expand()` help us
         // identify those cases.
    -    if (is_zero(existing_coeff.Expand() + coeff.Expand())) {
    -      map->erase(it);
    -    } else {
    +    //if (is_zero(existing_coeff.Expand() + coeff.Expand())) {
    +    //  map->erase(it);
    +    //} else {
           existing_coeff += coeff;
    -    }
    +    //}
       } else {
         // m ∉ dom(map)
         map->emplace_hint(it, m, coeff);
    ```
    it takes 12ms.

hongkai-dai, 2022-04-14:
@jwnimmer-tri since our last meeting, we discussed two ideas to improve the speed
1. When we call `ParseConstraint` with a symbolic expression, at the start of this function, we expand expression, and then just use the expanded expression afterwards.
2. In symbolic::Expression, store its expanded version after calling `Expand()` function.

I tried idea 1 in the dev [branch](https://github.com/hongkai-dai/drake/tree/expand_before_parse_constraint) but the benchmark speed on `solvers/benchmarking/benchmark_mathematical_program.cc` is unchanged.

I profiled this benchmark test. [This](https://www.dropbox.com/s/ehod1b7drt5thv0/callgrind.out.expand_expression?dl=0) is the callgrind result. Specifically it shows that computing the product `p_i.dot(q_i)` between polynomials takes 80% of time on this line https://github.com/RobotLocomotion/drake/blob/6e6e37ffa677362245773f13c0628f0042b47414/solvers/benchmarking/benchmark_mathematical_program.cc#L34
and calling AddSosConstraint takes the remaining 20% of time.

Moreover, kcachegrind thinks calling Expand() in these lines are the culprit https://github.com/RobotLocomotion/drake/blob/2ad76f8ca579bc2883faf6d7331cae4bb57256fe/common/symbolic_polynomial.cc#L46-L50

So I think to improve the speed, we really need to try idea 2, to store the expanded expression.

jwnimmer-tri, 2022-04-14:
I can confirm on my computer that only ~20% of time is spent in `AddConstraint`, and that ~80% is spent preparing the `Expression` that's passed into `AddConstraint`.  I'm working on improving my understanding now, and will ping back here with any observations and advice.

jwnimmer-tri, 2022-04-15:
WIP branch is at https://github.com/jwnimmer-tri/drake/commits/sym-sos-fast.  I've solve the costly `Expand()` problem, runtime is down from ~10s to ~3.4s now.  There's still more work that could be done to make coefficient accumulation faster, but I think we should work on merging the Expand fixups to master first.

hongkai-dai, 2022-07-27:
I will close this issue as `Expand()` function is a lot faster (and we avoided calling it in many cases).

closed 2022-07-27 by hongkai-dai