RobotLocomotion/drake #16300: Simplify ngrok authentication in StartMeshcat

RussTedrake, 2022-01-03:
Unfortunately, ngrok has changed their policies and broken my [current workflow](https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/_geometry_extra.py). See: https://github.com/RussTedrake/underactuated/issues/454
Now users must register, and supply their authtoken even to use the free service.
https://ngrok.com/docs#getting-started-authtoken

It would be tempting to propose:
```
def StartMeshcat(ngrok_auth_token=None)
```
and maybe it's ok to offer that, but I don't want that to be the primary workflow.  It would mean that any code that calls StartMeshcat with the token would not be immediately shareable, and I would have to continuously tiptoe around not accidentally sharing my token when I distribute the code.  Students will have to do the same.

I think it's better to support the workflow where people supply the token via a configuration file.  They could manually upload it to the colab/deepnote instance every time (yuck).  On colab, we could have the user be able to upload that token once in their google drive, and [we pull it from there](https://colab.research.google.com/notebooks/io.ipynb#scrollTo=Mounting_Google_Drive_locally), but that means that users have to grant permissions to our code to read their google drive auth file (yuck).  I haven't looked closely, but the example notebook on mounting the drive grants all read/write permissions to the entire google drive all at once; hopefully we could find something more narrow.

I don't see any good solutions here, so I guess I'll probably restart my investigation of ngrok alternatives:
https://github.com/RussTedrake/manipulation/issues/146

I wanted to open this issue to see if the drake devs have any ideas.

RussTedrake, 2022-01-03:
FYI @alexdlaird , in case you hadn't hit the unregistered connection change yet.

alexdlaird, 2022-01-03:
@RussTedrake Thanks for the heads up, I'll take a look!

jwnimmer-tri, 2022-02-10:
For Deepnote, I think we should change `StartMeshcat` to behave differently in case the easy-to-use 8080 default was already taken:

(1) Tell the user what's happening.  Explain how to close and/or restart the old notebooks that are camping on the port.  They might choose to do that, instead of fussing with ngrok.

(2) Then, also immediately prompt the user to paste in their ngrok token (along with instructions about how to make one).
```python
ngrok_auth_token = input()
```

If they type in nothing, then raise the "Could not open port" exception so the cell retires.  If they provide a token, then proceed with the ngrok setup.

When they provide a token, store it token in a notebook global variable, and in case we ever run `StartMeshcat()` again within the same interpreter, re-use the token instead of re-prompting.

---

A future improvement would be to ask Deepnote to add [keyring](https://pypi.org/project/keyring/) support to their servers, so that users could store user-specific secrets associated with their accounts, but that would not be shared with other users who run (or copy) those notebooks.  That's probably the best long-term answer.  Separate configuration files will be annoying.  The `keyring` API is the proper place for secret management via Python.

jwnimmer-tri, 2022-05-03:
@RussTedrake I wonder if we should close this issue now?  We don't use ngrok for our tutorials anymore.  It's an interesting topic, but possibly not a Drake issue?

closed 2022-05-03 by RussTedrake