RobotLocomotion/drake #16301: Upgrade externals to latest - January 2022

BetsyMcPhail, 2022-01-04:
Follow [Semi-automated monthly upgrades](https://github.com/RobotLocomotion/drake/blob/master/tools/workspace/README.md#semi-automated-monthly-upgrades) documentation

BetsyMcPhail, 2022-01-06:
For the most part, the script ran smoothly. I have questions about a few packages:

- `petsc needs upgrade from v3.16.1 to xsdk-0.3.0-rc1` This should `v3.16.3` ([github repo](https://github.com/petsc/petsc) for reference). For this month, I was able to work around this and manually update to `v3.16.3`.

- `ignition_math needs upgrade from ignition-math6_6.9.2 to ignition-math6_6.9.3-pre1` Do we support `pre` releases? ([github repo](https://github.com/ignitionrobotics/ign-math) for reference)

- `qhull needs upgrade from 2020.2 to v8.1-alpha1` ([github repo](https://github.com/qhull/qhull) for reference). I need to look into this some more, it looks like there's 2 versioning schemes? Again,

jwnimmer-tri, 2022-01-06:
You're right that we should not upgrade to alpha, beta, rc, etc. releases.  To the extent practical, we should add special-case logic to the `new_release.py` to fix its suggestions for these cases.

Yes, qhull has two version schemes.  I wasn't able to tell whether / how they differ.  In `new_release` we could have a heuristic like "if the name is 'qhull', then ignore tags that start with 'v'" or something.

BetsyMcPhail, 2022-01-06:
Both of those suggestions sound reasonable. I can take a pass at updating `new_release.py`. Should we make a similar change for petsc, only consider tags that start with 'v'?

jwnimmer-tri, 2022-01-06:
Sure.  Any heuristics that make the tool give better answers sound great.

BetsyMcPhail, 2022-01-06:
`stduuid needs upgrade from 4959d46eb494f8eddd4a37a1c8c6434b41d56ca8 to 9357e5280c1df9ad6f88ff9581464c8ae4bd7b50` resulted in build errors. For example, [linux-bionic-gcc-bazel-experimental-release](https://drake-jenkins.csail.mit.edu/view/Experimental/job/linux-bionic-gcc-bazel-experimental-release/13184/).

jwnimmer-tri, 2022-01-11:
Re: `stduuid`.  In general when upstream is broken, we should either file a bug report, or maybe even submit a pull request with the fix.  (This is one of the key outcomes of doing the monthly upgrade -- being able to keep pace with changes to upstream and offer feedback.)

jwnimmer-tri, 2022-01-14:
On first glance, it looks like https://github.com/mariusbancila/stduuid/pull/56 is the fix that we'll need.

BetsyMcPhail, 2022-01-14:
That looks promising. I'll give it a try.

jwnimmer-tri, 2022-01-14:
I _think_ everything here is finished now?  Or at least, I'm okay with calling it done and picking up any loose ends when the Feb 1st upgrades roll around.

BetsyMcPhail, 2022-01-14:
Everything has been updated. Closing issue for this month

closed 2022-01-14 by BetsyMcPhail