RobotLocomotion/drake #16307: Compute centroidal momentum matrix A(q) together with Adot(q, v) * v

hongkai-dai, 2022-01-05:
As requested in this stackoverflow question https://stackoverflow.com/questions/70576914/how-to-obtain-centroidal-momentum-matrix-explicitly-for-momentum-based-control-p

The centroidal momentum can be written as A(q) * v, where A(q) is called the centroidal momentum matrix. Currently we compute the centroidal momentum directly. But in controller design, it often needs to compute the matrix A(q) together with the term Adot(q, v) * v.

mitiguy, 2022-01-06:
Related issue #10779.
Paul's notes from face-to-face meeting with @hongkai-dai .
L, a system's spatial momentum about its center of mass can be written as
L = A(q) * v  where A(q) is a 6 x nv matrix (not square) where nv is the number of velocities.
L̇ = Ȧ v + A v̇,  where L̇  denotes the time-derivative of L in world W.
Although L is useful for motion planning, the matrix A(q) is useful for controls.
Drake needs to calculate A(q), called "the centroidal momentum matrix" explicitly (without AutoDiff).
Also needed is the term Ȧ v   (no need to calculate the individual factors Ȧ separately).

Paul's early thoughts on how to resolve this.
1.  Review code in MultibodyPlant::CalcSpatialMomentumInWorldAboutPoint().
   Vector3<T> p_WoScm_W =  plant.CalcCenterOfMassPositionInWorld(context);
   SpatialMomentum<T> L_WScm_W =  plant.CalcSpatialMomentumInWorldAboutPoint(context, p_WoScm_W); 

2. To calculate the A(q) matrix, leverage 
   MultibodyPlant::CalcJacobianSpatialVelocity().
   To verify the calculation of A(q), test L - A(q)*v  = 0.

3.  Create a new method MultibodyTree::CalcMomentOfInertiaForceAboutPoint().
This may necessitate new methods to calculate SpatialAcceleration of individual bodies.

4.  Notice that L̇  = MultibodyTree::CalcMomentOfInertiaForceAboutPoint( Scm ),
where Scm is the system's center of mass.
Calculate Ȧ v  = L̇  -  A v̇  (which has similarities to CalcBiasSpatialAcceleration).

sherm1, 2022-01-06:
BTW assuming we can already calculate fast centroidal `L = A v` we can get any column of A by setting v to [1 0 0 ... 0], [0 1 0 0 ... 0], etc. and making multiple calls. That could be wrapped in a CalcCentroidalMomentumMatrix() API which could be implemented very easily. Later if it is a performance issue it could be reimplemented with a fancier algorithm.

sherm1, 2022-01-06:
Looks like we could use the same function to get `Ȧ v − L̇ = −A v̇` (supplying −v̇ to the L function instead of v). Then add L̇ calculated as Paul said above to get `Ȧ v`.

sherm1, 2022-01-06:
FYI For a not-quite-the-same example of calculating a matrix one column at a time by repeated calls in Drake using a modified Context, see ImplicitIntegrator::CalcForwardDiffJacobian() in [systems/analysis/implicit_integrator.cc:104](https://github.com/RobotLocomotion/drake/blob/master/systems/analysis/implicit_integrator.cc#L104).