RobotLocomotion/drake #16313: plot_system_graphviz should use pydot instead of matplotlib (at least in jupyter)

RussTedrake, 2022-01-06:
I've made this change in my course notes... it outputs and renders an svg instead of a grainy downsampled bitmap in matplotlib.
The diagram in dynamical_systems_tutorial is currently embarrasingly bad.

It would require adding `pydot` as a python requirement.
this may become obsolete if we fix #13874

jwnimmer-tri, 2024-03-17:
>  The diagram in dynamical_systems_tutorial is currently embarrasingly bad.

At least as of today, the figures in our tutorials look fine to me.

It's not clear to me exactly what the problem or call to action is here.  The `plot_system_graphviz` has used pydot since 2017.

Or did you mean the call to `plt.imshow`?  Maybe that used to do poorly, but it looks fine today.

Maybe we should close this?

RussTedrake, 2024-03-17:
I agree that original issue description is stale.  The dynamical_systems tutorial was changed to use pydot in https://github.com/RobotLocomotion/drake/commit/97b1531209002b269e910bc055c09a791e3dd2f8
and looks great.

However, if you replace the lines in `dynamical_systems.ipynb` on deepnote
```
display(SVG(pydot.graph_from_dot_data(
    diagram.GetGraphvizString(max_depth=2))[0].create_svg()))
```
with
```
from pydrake.all import plot_system_graphviz
plot_system_graphviz(diagram)
```
the the output is much worse.  (it is rasterized, and i don't believe it supports scrolling the same way, etc).

I think the issue should stay open, and yes the call to action is to have plot_system_graphviz use `IPython.display` instead of `plt.imshow`.

PR #22526 (2025-01-27): plot_graphviz uses ipython display with svg in notebooks
Resolves #16313

+@rpoyner-tri for feature review, please.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22526)
commit 7501b5fd (2025-01-27): plot_graphviz uses ipython display with svg in notebooks (#22526)

Resolves #16313
closed 2025-01-27 by sammy-tri