RobotLocomotion/drake #16342: `UnitInertia::SolidCapsule(...)`

RussTedrake, 2022-01-10:
We support the Capsule shape, but don't offer the helper method for computing it's unit inertia.  I need it for the mujoco parser.
https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_unit_inertia.html

I hope this is a quick request.  It should be simple!  @sherm1 -- could you please delegate so we can knock it out?

sherm1, 2022-01-10:
@xuchen, for a look at another part of the code can I interest you in adding the inertia properties for capsule and ellipsoid to UnitInertia? Russ would like this asap. LMK if you can't fit this in soon.

xuchenhan-tri, 2022-01-10:
Sure. I'll take a stab today, but it might be tight with RTO stuff and meetings. Tuesday for sure.

RussTedrake, 2022-01-10:
Perfect.  Thank you!

PR #16357 (2022-01-14): [multibody] Add unit inertia for capsule
Fixes #16342

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16357)
closed 2022-01-14 by EricCousineau-TRI